tpaviot/pythonocc-core #68: On GetHandle(self): TypeError: int() argument must be a string or a number, not 'PyCapsule'

sanchezgt, 2015-01-05:
Using py3.4 on windows
TypeError: int() argument must be a string or a number, not 'PyCapsule'
on
def GetHandle(self):
        return int(self._winId())

replace by

def GetHandle(self):
        ctypes.pythonapi.PyCapsule_GetPointer.restype = ctypes.c_void_p
        ctypes.pythonapi.PyCapsule_GetPointer.argtypes = [ctypes.py_object]
        return ctypes.pythonapi.PyCapsule_GetPointer(self.winId(), None)

tpaviot, 2015-01-05:
@sanchezgt which file is affected, pyqt4Display.py or pysideDisplay.py? Which PyQt version or PySIde version do u use ? running which script ?

jf---, 2015-01-08:
we need to know which GUI library your using and what verion of pythonocc you are one.
this is an older version of pythonocc, the current you can [find here](https://github.com/tpaviot/pythonocc-core/releases):

Why not try and install that newer version?
Its important that the [OCE library](https://github.com/tpaviot/oce/releases) you've got installed match.
So pythonocc-core 0.16 goes with OCE 0.16

D-T-F, 2015-04-14:
I'have the same Problem.    

display, start_display, add_menu, add_function_to_menu = init_display()
  File "...\Display\SimpleGui.py", line 164, in init_display
    win.canva.InitDriver()
  File "...\Display\pysideDisplay.py", line 79, in InitDriver
    self._display = OCCViewer.Viewer3d(self.GetHandle())
  File "...\Display\pysideDisplay.py", line 55, in GetHandle
    return int(self.winId())
TypeError: int() argument must be a string, a bytes-like object or a number, not 'PyCapsule'

Windows7 
Python 3.4.3
pythonOCC-0.16.0-win32-py34.exe (26Mb) (takes from http://www.pythonocc.org/download/ today)
GUI library:  PySide-1.2.2-cp34-none-win32.whl 
script: "Hello dumb box !" (http://www.pythonocc.org/quick-examples/hello-dumb-box/)

at the moment i've no ideas on how to fix the problem

Thanks.

tpaviot, 2015-07-16:
Fixed by commit f91cd0e546c403dacf34c64089ce9a0816c5df51

closed 2015-07-16 by tpaviot

yogeshhk, 2015-12-04:
Same problem, even today.

Downloaded latest pythonocc and trying to run a sample in Python 3.4 in IDLE.

jf---, 2015-12-08:
Hi @yogeshhk ,

Can you please inform us which version of PyRAPID you're running?
On which platform are you running?
How have you installed pythonocc-core?

yogeshhk, 2015-12-09:
I downloaded pythonOCC-0.16.0-win32-py34.exe from Download | pythonOCC for my Win 7 64bit computer. I had Python 3.4 32bit already present.
It got installed in C:\Python34\Lib\site-packages\OCC
Tried running HelloBox sample, but got error at init_display() line
I don't know if I have PyRAPID you are talking about. How to check?
When I installed pythonocc I suppose I should have pythonocc-core. How to check?

jf---, 2015-12-09:
Hi @yogeshhk , 

Can you read follow the discussion in #160 ?
This is most likely a version issue, and resolved in a later version of pythonocc.

yogeshhk, 2015-12-10:
I am not building from source at the moment. Wish to have all-inclusive windows exe, to install and run. I guess I should wait for a while till the said commit is in the latest exe. Right?

jf---, 2015-12-10:
nope, install using conda, there are windows packages here:
https://anaconda.org/DLR-SC/pythonocc-core/

yogeshhk, 2015-12-13:
I am getting:

C:\Users\kulkarni>conda install -c https://conda.anaconda.org/dlr-sc pythonocc-core
Fetching package metadata: ......
Solving package specifications: ..........
Error: Unsatisfiable package specifications.
Generating hint:
[      COMPLETE      ]|##################################################| 100%

Hint: 'pythonocc-core' has unsatisfiable dependencies (see 'conda info pythonocc-core')

Note that the following features are enabled:
  - vc14

jf---, 2015-12-14:
@yogeshhk thanks, this is insightful...
could you run "conda info pythonocc-core" please? Thanks!

yogeshhk, 2015-12-14:
C:\Users\kulkarni>conda info pythonocc-core
Fetching package metadata: ....
Error: No packages found in current win-64 channels matching: pythonocc-core

yogeshhk, 2016-01-26:
I have Python 3.4.3 and I just wish to have all inclusive pythonocc installer for Windows 7 64 bit. Possible?