tpaviot/pythonocc-core #70: TopoDS_Shape subclasses cannot be inherited from

jf---, 2015-01-07:
The following is a pretty serious regression... 
AFAIK its not possible to inherit any TopoDS_Shape decendents

This is because the TopoDS_Edge ( and TopoDS_Face and friends ) 
do not define a ( copy? ) constructor that takes a existing ( TopoDS_Edge here ) instance as argument

I dont see how this can be by-passed using OCC.TopoDS.topods.Edge or OCC.TopoDS.topods_Edge

That means that we can't inherit from TopoDS_Shape subclasses... 

```python
from OCC.BRepBuilderAPI import BRepBuilderAPI_MakeEdge
from OCC.TopoDS import topods_Edge, TopoDS_Edge
from OCC.gp import gp_Pnt

line = BRepBuilderAPI_MakeEdge(gp_Pnt(), gp_Pnt(100,0,0)).Edge()

class InheritEdge(topods_Edge):
    def __init__(self, edge):
        super(InheritEdge, self).__init__(edge)

"""

cannot inherit from:

Traceback (most recent call last):
  File "/Users/jelleferinga/GIT/pythonocc-core/src/addons/Utils/tst.py", line 8, in <module>
    class InheritEdge(topods_Edge):
TypeError: Error when calling the metaclass bases
    function() argument 1 must be code, not str
"""

class InheritEdge(TopoDS_Edge):
    def __init__(self, edge):
        # no constructor defined... 
        # TopoDS_Edge() just returns an uninitialized TopoDS_Edge...
        super(InheritEdge, self).__init__(edge)
```

jf---, 2015-01-07:
following the [OCC 6.8.1 docs](http://dev.opencascade.org/doc/refman/html/class_topo_d_s___edge.html#details), the TopoDS_Edge indeed does not define a constructor... 

I need to dig a little further...

jf---, 2015-01-07:
probably this class should now inherit from TopoDS_Shape 

So, I think initialization is done in a really different way now, [see the docs here](http://dev.opencascade.org/doc/refman/html/class_topo_d_s___builder.html)

Perhaps our implementation could look like this:

```python

class MyFancyEdge(TopoDS_Shape):
    def __init__(self, edge):
        super(InheritEdge, self).__init__()
        # method below not available... its protected
        TopoDS_Builder.MakeShape( self, edge )

```

jf---, 2015-01-07:
from OCC import BRep_Builder

another way to initalize a TopoDS_Edge from another instance is to use the [**UpdateEdge**](http://dev.opencascade.org/doc/refman/html/class_b_rep___builder.html
) method from the BRep_Builder class. That class is not currently wrapped...

```python

from OCC import BRep_Builder # does not exist

class InheritEdge(TopoDS_Edge):
    def __init__(self, edge, eps=1e-6):
        # no constructor defined...
        # TopoDS_Edge() just returns an uninitialized TopoDS_Edge...
        super(InheritEdge, self).__init__()
        BRep_Builder.UpdateEdge (self, eps ) 
````

jf---, 2015-01-07:
checking out the source for TopoDS_Builder::MakeShape has been insightful:

```c++

void TopoDS_Builder::MakeShape (TopoDS_Shape& S,
                                const Handle(TopoDS_TShape)& T) const
{
  S.TShape(T);
  S.Location(TopLoc_Location());
  S.Orientation(TopAbs_FORWARD);
}

```

The following seems to work out:

```python
from OCC.BRepBuilderAPI import BRepBuilderAPI_MakeEdge
from OCC.TopAbs import TopAbs_FORWARD
from OCC.TopLoc import TopLoc_Location
from OCC.TopoDS import TopoDS_Edge
from OCC.gp import gp_Pnt
from OCC.BRep import BRep_Tool

line = BRepBuilderAPI_MakeEdge(gp_Pnt(), gp_Pnt(100,0,0)).Edge()

class InheritEdge(TopoDS_Edge):
    def __init__(self, edge):
        # no constructor defined...
        # TopoDS_Edge() just returns an uninitialized TopoDS_Edge...
        super(InheritEdge, self).__init__()

        self.TShape(edge.TShape())
        self.Location(edge.Location())
        self.Orientation(edge.Orientation())

    def get_curve(self):
        return BRep_Tool.Curve(self)

ie = InheritEdge(line)
print ie.get_curve()

>>> (<OCC.Geom.Handle_Geom_Curve; proxy of <Swig Object of type 'Handle_Geom_Curve *' at 0x1062c8150> >, 0.0, 100.0)

```

tpaviot, 2015-01-08:
@jf--- The latter is certainly the best way to proceed. See PR #71

closed 2015-01-08 by jf---