tpaviot/pythonocc-core #75: travis build failing | taking too long?

jf---, 2015-01-18:
looks like gcc compilation is perhaps taking too long for travis to complete the build?
our GH page says "build failing" which is not the best of advertisements...

is simple only running the clang build an option / quick fix @tpaviot ?

```bash
target _TColGeom
[ 86%] Built target _TCollection

>>> g++: internal compiler error: Killed (program cc1plus)

Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.6/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/_TColStd.dir/src/SWIG_files/wrapper/TColStdPYTHON_wrap.cxx.o] Error 4
make[1]: *** [CMakeFiles/_TColStd.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 86%] Built target _TColgp
[ 86%] Built target _TDF
[ 86%] Built target _TDataStd
make: *** [all] Error 2
Timestamp
Wed Jan 14 16:36:44 UTC 2015
Traceback (most recent call last):
  File "run_tests.py", line 22, in <module>
    import core_wrapper_features_unittest
  File "/home/travis/build/tpaviot/pythonocc-core/test/core_wrapper_features_unittest.py", line 23, in <module>
    from OCC.Standard import Standard_Transient, Handle_Standard_Transient
ImportError: No module named OCC.Standard
The command "./.travis.build.sh" exited with 1.
Done. Your build exited with 1.
```

tpaviot, 2015-01-18:
It's not a question of time (oce takes much more time tom compile). It may be possible to fix this by decreasing the number of concurrent compilation tasks (e.g. make -j4 to make -j3).

tpaviot, 2015-01-18:
An attempt with commit 8bc8733b025e2d904a989b6c6855961a33360ce8. Wait an see...

jf---, 2015-01-18:
OK, I'm curious...

tpaviot, 2015-01-18:
It doesn't work, it seems there's a limit of 50mn. I remember I asked for an extra time for oce ! So let's increase the number of concurent tasks. I amend the commit.

tpaviot, 2015-01-18:
ok, it seems to be fixed.

closed 2015-01-18 by tpaviot

jf---, 2015-01-19:
amazing! great intuition and cool the travis guys gave additional time on OCE