tpaviot/pythonocc-core #76: SetPixelFormat failed. Error code: 2000

aothms, 2015-01-28:
Hi,

I get the following stacktrace when initializing SimpleGui, using the 0.16 release on Windows.

    File "C:\Program Files (x86)\Python2.7.9\lib\site-packages\OCC\Display\SimpleGui.py", line 164, in init_display
        win.canva.InitDriver()
      File "C:\Program Files (x86)\Python2.7.9\lib\site-packages\OCC\Display\pyqt4Display.py", line 80, in InitDriver
        self._display.Create()
      File "C:\Program Files (x86)\Python2.7.9\lib\site-packages\OCC\Display\OCCViewer.py", line 105, in Create
        self.Init(self._window_handle)
    RuntimeError: Aspect_GraphicDeviceDefinitionError
    OpenGl_Window::CreateWindow: SetPixelFormat failed. Error code: 2000
    
I looked around and saw you referenced the same error, some time ago:
https://mail.gna.org/public/pythonocc-users/2013-04/msg00010.html
(coincidently I am also on an acer notebook)

Is this something that is still on the radar? I saw a stackoverflow post with similar problems and nasty solutions:
http://stackoverflow.com/questions/199016/wglcreatecontext-in-c-sharp-failing-but-not-in-managed-c#answer-206933

Looking forward to your insights,

Thanks,
Thomas

tpaviot, 2015-01-28:
This has been a long time since this issue has not been reported! It used to be an usual issue some time go.

This issue comes from your graphic driver/graphic card. I had the same issue on a builtin Intel graphic chipset. Did you experiment similar behavior with an older pythonocc version ?

aothms, 2015-02-01:
Hi Thomas,

Thanks for your quick reply. Actually you are very right, I updated my graphics driver (indeed, also for a builtin intell chip) and it works flawlessly now. From my memory I did have similar problems on this same laptop with other pyocc versions, I can check if you want to. But since it's easily fixed by updating the driver I wouldn't spend much brain cycles on it.

Thanks again,

Best,
Thomas

closed 2015-02-01 by aothms