tpaviot/pythonocc-core #77: ASCII STL files lack space after "solid"

WoLpH, 2015-01-29:
One of the users of my Numpy-STL project (https://github.com/WoLpH/numpy-stl) reported that the files produced by PythonOCC don't (always?) work with Numpy-STL, he noticed that the ASCII STL files lack a space after "solid" which is officially required by the standard.

Or at least, the Wikipedia page about the STL standard claims it does: http://en.wikipedia.org/wiki/STL_(file_format)#ASCII_STL

Thanks to Konrad Schneider for reporting the bug.

tpaviot, 2015-01-29:
Thanks for the feedback. This comes from the underlying OCE library, which has to be fixed. However, the link provided is only a wikipedia entry, is there any link to any kind of official specification ?

WoLpH, 2015-01-29:
To be honest... can't find an official standard anywhere.
A few possibilities:

 - http://people.sc.fsu.edu/~jburkardt/data/stla/stla.html
 - http://www.ennex.com/~fabbers/StL.asp

tpaviot, 2015-07-15:
@WoLpH This issue has to be submitted to the oce project (https://github.com/tpaviot/oce/issues)

closed 2015-07-15 by tpaviot