tpaviot/pythonocc-core #80: STEP file reader issue.

dev1intellial, 2015-02-19:
I am trying to read STEP file using python. Read the installation document file from github and download all the necessary file 
( For example. 
- the python programming language (http://www.python.org). Python 2.7 and 3.x
 are officially supported,
- OpenCascade Community Edition 0.16 (https://github.com/tpaviot/oce),
- SWIG 2.0.10 or higher (http://www.swig.org),
- CMake 2.8 or higher (http://www.cmake.org).
)
but problem is that when I am going to use cmake command then I am getting the following error. I am new in python so I am not able to understand how to solve this problem. Is there any easy documentation to run this program means read step file using python. 

This is the error when running this cmake command..
D:\software\python\pythonocc-core-master\cmake-build>cmake ..
CMake Warning at CMakeLists.txt:18 (project):
  To use the NMake generator, cmake must be run from a shell that can use the
  compiler cl from the command line.  This environment does not contain
  INCLUDE, LIB, or LIBPATH, and these must be set for the cl compiler to
  work.
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:18 (project):
  The CMAKE_C_COMPILER:
    cl
  is not a full path and was not found in the PATH.
  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:18 (project):
The CMAKE_CXX_COMPILER:
    cl
  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "D:/software/python/pythonocc-core-master/CMakeFiles/CMakeOutput.log".
See also "D:/software/python/pythonocc-core-master/CMakeFiles/CMakeError.log".

Thanx in advance.

jf---, 2015-02-19:
What stops you from using the [binaries](https://github.com/tpaviot/pythonocc-core/releases)?

dev1intellial, 2015-02-19:
I want to run this program ..

from OCC.STEPControl import STEPControl_Reader
from OCC.Display.WebGl import x3dom_renderer
step_reader = STEPControl_Reader()
step_reader.ReadFile('cylinder_block.stp')
step_reader.TransferRoot()
block_cylinder_shape = step_reader.Shape()
# render cylinder head in x3dom
my_renderer = x3dom_renderer.X3DomRenderer()
my_renderer.DisplayShape(block_cylinder_shape)

but its showing ImportError: No module named 'OCC'

That's why I am not able to run this program. How to run this program?

jf---, 2015-02-19:
Did you install the module? Does your python version match the version of the installer?

dev1intellial, 2015-02-19:
How to get this "OCC" module?  I just download the node-occ master from Open CASCADE Community.

jf---, 2015-02-19:
this is pythonocc, not node-occc

dev1intellial, 2015-02-19:
Ok, I have already downloaded pythonocc.

dev1intellial, 2015-02-19:
Can you give some idea how to do that?

jf---, 2015-02-19:
Alas, nope... 
Keep us posted of your experiments ;)

tpaviot, 2015-07-15:
@dev1intellial  This is an issue related to the cmake program. Please check the documentation in order to use MS nmake with cmake.

closed 2015-07-15 by tpaviot

Abhinav-Anil, 2020-03-03:
Can we do parser through pythonocc??

tpaviot, 2020-03-03:
This issue was closed 5 years ago. Please open a new issue