tpaviot/pythonocc-core #82: make error Building CXX object CMakeFiles/_AIS.dir/src/SWIG_files/wrapper/AISPYTHON_wrap.cxx.o

Lionheard, 2015-02-24:
Hello
i tryed many times to get Pythonocc run but every time the same issue. Here isthe Terminal output:

paul@paul-MS-7748:~/Downloads/Swig/PythonOCC/pythonocc-core-0.16.0$ sudo make
1) [  0%] Building CXX object CMakeFiles/_AIS.dir/src/SWIG_files/wrapper/AISPYTHON_wrap.cxx.o
2) /home/paul/Downloads/Swig/PythonOCC/pythonocc-core-0.16.0/src/SWIG_files/wrapper
3) /AISPYTHON_wrap.cxx:4584:32: fatal error: Standard_Failure.hxx: Datei oder Verzeichnis nicht gefunden
 #include <Standard_Failure.hxx>
                                ^
compilation terminated.
make[2]: *** [CMakeFiles/_AIS.dir/src/SWIG_files/wrapper/AISPYTHON_wrap.cxx.o] Fehler 1
make[1]: *** [CMakeFiles/_AIS.dir/all] Fehler 2
make: *** [all] Fehler 2
In Line 1 the file wrapper is empty and in Line 2 and 3 there is the file AISPYTHON_wrap.cxx without where the Build looks at the same file with a .o at the end.
On myself iam not able to get working please can anybody help me?

If there are further questions what you need to know please let me know.

Many thanks to you

tpaviot, 2015-02-26:
You have to define the OCE include and library path.

Lionheard, 2015-02-26:
Hello tpaviot
Thank you very much fore your quick answer.
But iam a Dummy and have no idea how to define the OCE include and library path.
If it is possible to do this in a terminal please be so kind and give me a step by step instruction because that i just can handle. OCE has many many files.
Thanks a lot fore your helpness.

tpaviot, 2015-02-26:
Check https://github.com/tpaviot/pythonocc-core/blob/master/INSTALL.md

closed 2015-06-20 by tpaviot