tpaviot/pythonocc-core #83: STEP file reading issue in Python.

dev1intellial, 2015-02-25:
I am using Python3.4.2 and pythonOCC-0.16.0-win32-py34.exe to draw components. Every components are rendered properly with one defined color but that is not look like a real world component.

[image]

Above image is my Python implementation which generate 3D image from STEP file with one color.

Below image is rendered one of my windows software and there I have used Step file. I want to render component same as look like in below image so its look like a real world component.

[image]

Is there any way to get correct colored output in Python by read STEP file? I have searched a lot but didn't get a way to implement it. Please help me to go in forward direction.

from __future__ import print_function
import sys
#from OCC.STEPCAFControl import STEPCAFControl_Reader
from OCC.STEPControl import STEPControl_Reader
from OCC.IFSelect import IFSelect_RetDone, IFSelect_ItemsByEntity
from OCC.Display.SimpleGui import init_display
from OCC.Display.WebGl import threejs_renderer

step_reader = STEPControl_Reader()
status = step_reader.ReadFile('./models/test.STEP')

if status == IFSelect_RetDone:  # check status
    failsonly = False
    step_reader.PrintCheckLoad(failsonly, IFSelect_ItemsByEntity)
    step_reader.PrintCheckTransfer(failsonly, IFSelect_ItemsByEntity)

    ok = step_reader.TransferRoot(1)
    _nbs = step_reader.NbShapes()
    aResShape = step_reader.Shape(1)
else:
    print("Error: can't read file.")
    sys.exit(0)
   
#display, start_display, add_menu, add_function_to_menu = init_display()
#display.DisplayShape(aResShape, update=True)
#start_display()

my_renderer = threejs_renderer.ThreejsRenderer(background_color="#123345")
my_renderer.DisplayShape(aResShape)
           
The above code is used for read STEP file using Python.

Priyank30, 2015-02-27:
@dev1intellial @jf--- @tpaviot 
I am too facing exactly same issue as in described question. Does pythonocc supports more then one color for object if its defined in step file? 
So far I have tested number of objects, and all have same color where originally they do have more then one color.

tpaviot, 2015-02-27:
Yes, OCC supports import colors into STEP files, so pythonocc does. Hey guys, what's the matter with exporting color using STEP? honestly, who cares ?

dev1intellial, 2015-02-27:
Thanx for reply. Can you give one example which will display object with exact color?

dev1intellial, 2015-03-03:
Can you give some idea?

jf---, 2015-07-28:
@tpaviot , lets close the issue...

tpaviot, 2015-07-31:
@dev1intellial Please post this thread to the pythonocc google groups if you still fail to get it work.

closed 2015-07-31 by tpaviot