tpaviot/pythonocc-core #84: Catching an exception produces memory corruption and core dump

ambrusc, 2015-02-26:
This is against:
__OCE-0.16__ (https://github.com/tpaviot/oce/tree/OCE-0.16)
__pythonocc-core 0.16.0__ (https://github.com/tpaviot/pythonocc-core/tree/0.16.0)

I noticed that `Geom2dAPI_InterCurveCurve::Segment(...)` is wrapped such that the user has to pass in new handles.

When the exception is triggered in `::Segment(...)` the curves assigned to them are probably released in C++ and I suspect python is also deallocating them during garbage collection.

> The fact that there *is* an exception in `::Segment(...)` is probably an OCE bug

That's my guess anyway.

_This appears to prevent the use of the `Segment` method because it could blow your program up at any moment!_

Any help is much appreciated.

Here's the example code:
```python
# Ambrus Csaszar 02/25/2015

from OCC import Geom2d
from OCC import Geom2dAPI
from OCC import gp

def build_line2d(origin, direction, t1, t2):
    gp_origin = gp.gp_Pnt2d(gp.gp_XY(*origin))
    gp_direction = gp.gp_Dir2d(gp.gp_XY(*direction))
    curve2d = Geom2d.Geom2d_Line(gp_origin, gp_direction)
    return Geom2d.Geom2d_TrimmedCurve(Geom2d.Handle_Geom2d_Curve(curve2d), t1, t2)

def isect2d(pcurve1, pcurve2, tol):
    # Create handles
    hpcurve1 = Geom2d.Handle_Geom2d_Curve(pcurve1)
    hpcurve2 = Geom2d.Handle_Geom2d_Curve(pcurve2)
    # Create a 2-curve intersector
    sector = Geom2dAPI.Geom2dAPI_InterCurveCurve(hpcurve1, hpcurve2, tol)
    # Count intersections
    segment_count = sector.NbSegments()
    segments = []

    # Extract segment intersections
    for i in xrange(segment_count):

        # We probably shouldn't allocate these in python
        result1 = Geom2d.Handle_Geom2d_Curve()
        result2 = Geom2d.Handle_Geom2d_Curve()

        # This could probably just return:
        #   * a single handle if this is a curve-self intersector
        #   * a tuple of two handles if this is a curve-curve intersector
        sector.Segment(i + 1, result1, result2)

        result1 = None if result1.IsNull() else result1.GetObject()
        result2 = None if result2.IsNull() else result2.GetObject()

        segments.append((result1, result2))

    return segments

if __name__ == "__main__":

    params1 = {
        'origin': [2.922649714250443, -2.5120000000000005],
        'direction': [-0.49999999326941136, -0.8660254076703459],
        't1': 0.0,
        't2': 0.10392304798782465,
    }
    params2 = {
        'origin': [2.7949999993942467, -2.7330957956708906],
        'direction': [0.4999999932694151, 0.8660254076703436],
        't1': 0.0,
        't2': 0.1513763851611979,
    }

    pcurve1 = build_line2d(**params1)
    pcurve2 = build_line2d(**params2)

    tol = 1e-10

    for x in xrange(2):

        # Catching the standard construction error here results in a
        # double-free + core dump
        try:
            isect2d(pcurve1, pcurve2, tol)
        except Exception as ex:
            print ex.message
```

ambrusc, 2015-03-10:
Any thoughts on this? Seems like it might be an easy-enough fix in the wrapper.

tpaviot, 2015-07-15:
I can' reproduce the issue on my Ubuntu 14.04 machine, running latest pythonocc-0.16.1. Please @ambrusc reopen the issue if ever the problem is still there.

closed 2015-07-15 by tpaviot