tpaviot/pythonocc-core #86: ImportError: dynamic module does not define init function

scicalculator, 2015-02-27:
I installed pythonocc-core in a virtualenvironment using python 3.4 with is Ubuntu 14.04. No errors came up. I attempted to run the example code from: 

http://www.pythonocc.org/quick-examples/hello-dumb-box/

Running the code as is produces the following error. It seems to be attempting to import c-modules that are not designed to be imported. Any thoughts on what I have done wrong.

```
Traceback (most recent call last):
  File "hello.py", line 2, in <module>
    from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox
  File "/home/derek/.virtualenvs/occ/lib/python3.4/site-packages/OCC/BRepPrimAPI.py", line 32, in <module>
    _BRepPrimAPI = swig_import_helper()
  File "/home/derek/.virtualenvs/occ/lib/python3.4/site-packages/OCC/BRepPrimAPI.py", line 28, in swig_import_helper
    _mod = imp.load_module('_BRepPrimAPI', fp, pathname, description)
  File "/home/derek/.virtualenvs/occ/lib/python3.4/imp.py", line 243, in load_module
    return load_dynamic(name, filename, file)
ImportError: dynamic module does not define init function (PyInit__BRepPrimAPI)
```

tpaviot, 2015-07-15:
@scicalculator Be sure OCE libraries are in your PATH

jf---, 2015-07-28:
@tpaviot I think this can be closed...

tpaviot, 2015-07-30:
@scicalculator Please test again with latest pythonocc (0.16.2) release.

closed 2015-07-30 by tpaviot