tpaviot/pythonocc-core #88: Problem with boolean operation

Achim66, 2015-03-05:
Hello,
I'm just working with BSplineSurfaces, Solids and boolean operations.
In the following example the BRepAlgoAPI_Cut and BRepAlgoAPI_Common functions don't work.
The BRepAlgoAPI_Fude function is working.

#######################################################
from OCC.gp import *
from OCC.TColgp import *
from OCC.TColStd import *
from OCC.Geom import *
from OCC.GeomFill import *
from OCC.BRepBuilderAPI import *
from OCC.BRepFill import *
from OCC.GC import *
from OCC.TopoDS import *
from OCC.BRep import *
from OCC.BRepPrimAPI import *
from OCC.BRepAlgoAPI import *

from OCC.Display.SimpleGui import init_display
display, start_display, add_menu, add_function_to_menu = init_display()
display.DisableAntiAliasing()
def bsplineSurface(zVal,xFak_Val,yFak_Val):
    nCP_U = 4
    nCP_V = 4
    degree_U = 3
    degree_V = 3
    #k_U = degree_U +1
    #k_V = degree_V +1
    z = zVal
    xFak = xFak_Val
    yFak = yFak_Val
    array_Pol = TColgp_Array2OfPnt(1, nCP_U,1,nCP_V)
    array_Pol.SetValue(1,1, gp_Pnt(0*xFak, 0*yFak, 0))
    array_Pol.SetValue(1,2, gp_Pnt(0*xFak, 1*yFak, 0))
    array_Pol.SetValue(1,3, gp_Pnt(0*xFak, 2*yFak, 0))
    array_Pol.SetValue(1,4, gp_Pnt(0*xFak, 3*yFak, 0))
    array_Pol.SetValue(2,1, gp_Pnt(1*xFak, 0*yFak, 0))
    array_Pol.SetValue(2,2, gp_Pnt(1*xFak, 1*yFak, z))
    array_Pol.SetValue(2,3, gp_Pnt(1*xFak, 2*yFak, z))
    array_Pol.SetValue(2,4, gp_Pnt(1*xFak, 3*yFak, 0))
    array_Pol.SetValue(3,1, gp_Pnt(2*xFak, 0*yFak, 0))
    array_Pol.SetValue(3,2, gp_Pnt(2*xFak, 1*yFak, z))
    array_Pol.SetValue(3,3, gp_Pnt(2*xFak, 2*yFak, z))
    array_Pol.SetValue(3,4, gp_Pnt(2*xFak, 3*yFak, 0))
    array_Pol.SetValue(4,1, gp_Pnt(3*xFak, 0*yFak, 0))
    array_Pol.SetValue(4,2, gp_Pnt(3*xFak, 1*yFak, 0))
    array_Pol.SetValue(4,3, gp_Pnt(3*xFak, 2*yFak, 0))
    array_Pol.SetValue(4,4, gp_Pnt(3*xFak, 3*yFak, 0))
    array_Knots_U = TColStd_Array1OfReal(1, 2)
    array_Knots_U.SetValue(1, 0.0)
    array_Knots_U.SetValue(2, 0.1)
    array_Multi_U = TColStd_Array1OfInteger(1,2)
    array_Multi_U.SetValue(1,4)
    array_Multi_U.SetValue(2,4)
    array_Knots_V = TColStd_Array1OfReal(1,2)
    array_Knots_V.SetValue(1, 0.0)
    array_Knots_V.SetValue(2, 0.1)
    array_Multi_V = TColStd_Array1OfInteger(1,2)
    array_Multi_V.SetValue(1,4)
    array_Multi_V.SetValue(2,4)
    return Geom_BSplineSurface(array_Pol,array_Knots_U,array_Knots_V,array_Multi_U,array_Multi_V,degree_U,degree_V,False,False)

if __name__ == '__main__':
    aTol = 1e-06
    aSurface =bsplineSurface(2,1,1)
    aFace = BRepBuilderAPI_MakeFace(aSurface.GetHandle(),aTol)
    aSurface1 = bsplineSurface(-4,1,1)
    aFace1 = BRepBuilderAPI_MakeFace(aSurface1.GetHandle(),aTol)
    aBuilder = BRep_Builder()
    aShell = TopoDS_Shell()
    aBuilder.MakeShell(aShell)
    aBuilder.Add(aShell,aFace.Shape())
    aBuilder.Add(aShell,aFace1.Shape())
    aSolid = BRepBuilderAPI_MakeSolid(aShell)
    anOrigin = gp_Pnt(1.5,1.5,-5)
    aDir = gp_Dir(gp_XYZ(0,0,1))
    anAxis = gp_Ax2(anOrigin,aDir)
    aRadius = 0.5
    aHeight = 10
    aCylinder = BRepPrimAPI_MakeCylinder(anAxis,aRadius,aHeight)
    aCut = BRepAlgoAPI_Cut(aSolid.Shape(),aCylinder.Shape())
    aCommon = BRepAlgoAPI_Common(aSolid.Shape(),aCylinder.Shape())
    aFuse = BRepAlgoAPI_Fuse(aSolid.Shape(),aCylinder.Shape())
    display.DisplayShape(aCut.Shape())
    #display.DisplayShape(aCommon.Shape())
    #display.DisplayShape(aFuse.Shape())
    display.FitAll()
    start_display()
##############################################################

Some ideas?
Best Regards
Achim

jf---, 2015-03-05:
hi Achim,

Just glancing your code, I think you might forgot to call aCut.Perform() ?
A pattern is also to check the .IsDone method

Achim66, 2015-03-06:
Hello,
I exported the solid made out of the BSplineSurfaces as step-file, imported it in GAMBIT (GAMBIT is a former modelling and meshing tool of FLUENT respectivley ANSYS, ACIS-Kernel version 16), exported it there as step-file again and reimported it in pythonocc. In pythonocc i made the cut with the cylinder ... and it worked. So I think this problem should be solved by OpenCascade team - in my opinion.
Best regards
Achim

Achim66, 2015-03-06:
Hi Jell,
thank you for replying.
the Perform method doesn*t exist.

aCut.Perform()
AttributeError: 'BRepAlgoAPI_Cut' object has no attribute 'Perform'

the IsDone() Method results in True.

tpaviot, 2015-04-01:
@Achim66 I ran your code, it runs. Below is what I get, is it what is expected?
[image]

Achim66, 2015-04-01:
@tpaviot Thank you for replying.
[image]
I want to substract the cylinder from the other shape. So there should be an hole inside the shape made out of two bspline surfaces. It worked by exporting the bspline shape as step file, importing it in GAMBIT, exporting it again as step file, importing it in pythonocc and then perform the cut operation.

Achim66, 2015-04-01:
@tpaviot 
Thomas, I repeated the steps as i wrote before.
I imported the pythonocc step file and then made the cut.
And I imported the GAMBIT step file and made the cut.
Two different results:
- the pythonocc file was treates as a shell
- the GAMBIT file was treates as a solid
[image]
pythonocc step file - cut result
[image]
GAMBIT step file - cut result

Here is the code for export and import:

from OCC.STEPControl import *
my_step_exporter = STEPControl_Writer()
my_step_exporter.Transfer(aSolid.Shape(),STEPControl_AsIs)
my_step_exporter.Write("TestShape.stp")
step_reader = STEPControl_Reader()
#step_reader.ReadFile("TestShapeGambit.stp")
step_reader.ReadFile("TestShape.stp")
step_reader.TransferRoot(1)
anImportedShape = step_reader.Shape(1)
aCut1 = BRepAlgoAPI_Cut(anImportedShape,aCylinder.Shape())
display.DisplayShape(aCut1.Shape())

here ist the GAMBIT step file:

ISO-10303-21;
HEADER;
FILE_DESCRIPTION( ( 'STEP AP214' ), '1' );
FILE_NAME( '/home/achim/TestShapeGambit.stp', '  ', ( '  ' ), ( '  ' ), 'XStep 1.0', '  ', '  ' );
FILE_SCHEMA( ( 'automotive_design' ) );
ENDSEC;
DATA;
#1 = MECHANICAL_DESIGN_GEOMETRIC_PRESENTATION_REPRESENTATION( ' ', ( #10, #11, #12, #13, #14, #15, #16, #17, #18, #19 ), #6 );
#2 = PRODUCT_DEFINITION_CONTEXT( '', #20, 'design' );
#3 = APPLICATION_PROTOCOL_DEFINITION( 'INTERNATIONAL STANDARD', 'automotive_design', 1994, #20 );
#4 = PRODUCT_CATEGORY_RELATIONSHIP( 'NONE', 'NONE', #21, #22 );
#5 = SHAPE_DEFINITION_REPRESENTATION( #23, #24 );
#6 =  ( GEOMETRIC_REPRESENTATION_CONTEXT( 3 )GLOBAL_UNCERTAINTY_ASSIGNED_CONTEXT( ( #27 ) )GLOBAL_UNIT_ASSIGNED_CONTEXT( ( #29, #30, #31 ) )REPRESENTATION_CONTEXT( 'NONE', 'WORKSPACE' ) );
#10 = STYLED_ITEM( '', ( #33 ), #34 );
#11 = STYLED_ITEM( '', ( #35 ), #36 );
#12 = STYLED_ITEM( '', ( #37 ), #38 );
#13 = STYLED_ITEM( '', ( #39 ), #40 );
#14 = STYLED_ITEM( '', ( #41 ), #42 );
#15 = STYLED_ITEM( '', ( #43 ), #44 );
#16 = STYLED_ITEM( '', ( #45 ), #46 );
#17 = STYLED_ITEM( '', ( #47 ), #48 );
#18 = STYLED_ITEM( '', ( #49 ), #50 );
#19 = STYLED_ITEM( '', ( #51 ), #52 );
#20 = APPLICATION_CONTEXT( ' ' );
#21 = PRODUCT_CATEGORY( 'part', 'NONE' );
#22 = PRODUCT_RELATED_PRODUCT_CATEGORY( 'detail', ' ', ( #53 ) );
#23 = PRODUCT_DEFINITION_SHAPE( 'NONE', 'NONE', #54 );
#24 = ADVANCED_BREP_SHAPE_REPRESENTATION( 'volume.1', ( #55, #56 ), #6 );
#27 = UNCERTAINTY_MEASURE_WITH_UNIT( LENGTH_MEASURE( 1.00000000000000E-06 ), #29, '', '' );
#29 =  ( CONVERSION_BASED_UNIT( 'MILLIMETRE', #59 )LENGTH_UNIT(  )NAMED_UNIT( #62 ) );
#30 =  ( NAMED_UNIT( #64 )PLANE_ANGLE_UNIT(  )SI_UNIT( $, .RADIAN. ) );
#31 =  ( NAMED_UNIT( #64 )SOLID_ANGLE_UNIT(  )SI_UNIT( $, .STERADIAN. ) );
#33 = PRESENTATION_STYLE_ASSIGNMENT( ( #70 ) );
#34 = EDGE_CURVE( 'edge.2', #42, #52, #71, .T. );
#35 = PRESENTATION_STYLE_ASSIGNMENT( ( #72 ) );
#36 = ADVANCED_FACE( 'face.1', ( #73 ), #74, .F. );
#37 = PRESENTATION_STYLE_ASSIGNMENT( ( #75 ) );
#38 = EDGE_CURVE( 'edge.4', #52, #44, #76, .T. );
#39 = PRESENTATION_STYLE_ASSIGNMENT( ( #77 ) );
#40 = VERTEX_POINT( 'vertex.2', #78 );
#41 = PRESENTATION_STYLE_ASSIGNMENT( ( #79 ) );
#42 = VERTEX_POINT( 'vertex.1', #80 );
[... 107 lines cut ...]
#160 = COLOUR_RGB( '', 0.500000000000000, 0.500000000000000, 0.500000000000000 );
#161 = CARTESIAN_POINT( '', ( 0.00000000000000, 0.00000000000000, 0.00000000000000 ) );
#162 = CARTESIAN_POINT( '', ( 1.00000000000000, 0.00000000000000, 0.00000000000000 ) );
#163 = CARTESIAN_POINT( '', ( 2.00000000000000, 0.00000000000000, 0.00000000000000 ) );
#164 = CARTESIAN_POINT( '', ( 3.00000000000000, 0.00000000000000, 0.00000000000000 ) );
#165 = PRE_DEFINED_MARKER( '' );
#166 = COLOUR_RGB( '', 0.500000000000000, 0.500000000000000, 0.500000000000000 );
#168 = SURFACE_STYLE_FILL_AREA( #178 );
#169 = ORIENTED_EDGE( '', *, *, #34, .T. );
#170 = ORIENTED_EDGE( '', *, *, #38, .T. );
#171 = ORIENTED_EDGE( '', *, *, #46, .F. );
#172 = ORIENTED_EDGE( '', *, *, #50, .F. );
#173 = SURFACE_STYLE_FILL_AREA( #179 );
#174 = ORIENTED_EDGE( '', *, *, #34, .F. );
#175 = ORIENTED_EDGE( '', *, *, #50, .T. );
#176 = ORIENTED_EDGE( '', *, *, #46, .T. );
#177 = ORIENTED_EDGE( '', *, *, #38, .F. );
#178 = FILL_AREA_STYLE( '', ( #180 ) );
#179 = FILL_AREA_STYLE( '', ( #181 ) );
#180 = FILL_AREA_STYLE_COLOUR( '', #182 );
#181 = FILL_AREA_STYLE_COLOUR( '', #183 );
#182 = COLOUR_RGB( '', 0.500000000000000, 0.500000000000000, 0.500000000000000 );
#183 = COLOUR_RGB( '', 0.500000000000000, 0.500000000000000, 0.500000000000000 );
ENDSEC;
END-ISO-10303-21;

jf---, 2015-07-28:
@tpaviot , lets close the issue?

closed 2015-07-29 by Achim66

jf---, 2015-07-29:
thanks Achim...

monkpit, 2015-12-29:
@Achim66 just FYI you should probably edit this comment to use code formatting, otherwise on github it looks like you're referencing every issue and pull request with #... :)