tpaviot/pythonocc-core #90: Missing files in swig-wrapper during compilation

scicalculator, 2015-03-11:
I am building pythonocc-core on a CentOS 7 virtual machine. First, I built the `tpaviot/oce` repo without problems. The pythonocc-core installation had the following after running `make`. I tried this using both the latest github repo and the 0.16.0 release.

```
[  0%] [  0%] [  0%] Building CXX object CMakeFiles/_AdvApp2Var.dir/src/SWIG_files/wrapper/AdvApp2VarPYTHON_wrap.cxx.o
Building CXX object CMakeFiles/_Adaptor2d.dir/src/SWIG_files/wrapper/Adaptor2dPYTHON_wrap.cxx.o
Building CXX object CMakeFiles/_Adaptor3d.dir/src/SWIG_files/wrapper/Adaptor3dPYTHON_wrap.cxx.o
[  0%] Building CXX object CMakeFiles/_AIS.dir/src/SWIG_files/wrapper/AISPYTHON_wrap.cxx.o
/home/derek/repo/pythonocc-core/build/src/SWIG_files/wrapper/AdvApp2VarPYTHON_wrap.cxx:3921:27: fatal error: Standard_OId.hxx: No such file or directory
 #include<Standard_OId.hxx>
                           ^
compilation terminated.
/home/derek/repo/pythonocc-core/build/src/SWIG_files/wrapper/Adaptor2dPYTHON_wrap.cxx:3804:27: fatal error: Standard_OId.hxx: No such file or directory
 #include<Standard_OId.hxx>
                           ^
compilation terminated.
/home/derek/repo/pythonocc-core/build/src/SWIG_files/wrapper/Adaptor3dPYTHON_wrap.cxx:3938:27: fatal error: Standard_OId.hxx: No such file or directory
 #include<Standard_OId.hxx>
                           ^
compilation terminated.
/home/derek/repo/pythonocc-core/build/src/SWIG_files/wrapper/AISPYTHON_wrap.cxx:4616:33: fatal error: AIS_ConnectedShape.hxx: No such file or directory
 #include<AIS_ConnectedShape.hxx>
                                 ^
compilation terminated.
make[2]: *** [CMakeFiles/_Adaptor2d.dir/src/SWIG_files/wrapper/Adaptor2dPYTHON_wrap.cxx.o] Error 1
make[2]: *** [CMakeFiles/_AdvApp2Var.dir/src/SWIG_files/wrapper/AdvApp2VarPYTHON_wrap.cxx.o] Error 1
make[1]: *** [CMakeFiles/_Adaptor2d.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/_AdvApp2Var.dir/all] Error 2
make[2]: *** [CMakeFiles/_Adaptor3d.dir/src/SWIG_files/wrapper/Adaptor3dPYTHON_wrap.cxx.o] Error 1
make[1]: *** [CMakeFiles/_Adaptor3d.dir/all] Error 2
make[2]: *** [CMakeFiles/_AIS.dir/src/SWIG_files/wrapper/AISPYTHON_wrap.cxx.o] Error 1
make[1]: *** [CMakeFiles/_AIS.dir/all] Error 2
make: *** [all] Error 2

```

I am uncertain for the missing `*.hxx` files are expected to be. I am first tried using swig 2.0.10 through the package manager. I also attempted to use swig 3.0.5 from swig.org. Both versions of swig had the same problem. 

Any help on how to finish this compilation or which dependency I am missing would be greatly appreciated. Thank you

scicalculator, 2015-03-12:
It looks like `Standard_OId.hxx` is a file in OCE 0.15 that has since been removed.

scicalculator, 2015-03-13:
I finally realized that the community edition of OCE had recently been updated to 0.17. Because I was using the git repo, I was using the wrong version. Changing over the the 0.16.1 tag fixed the compilation problems

closed 2015-03-13 by scicalculator

jf---, 2015-03-16:
Hi @scicalculator so yes, we did change / sync'd up the version names of OCE and pythonocc, since these *absolutely* need to match up. The SWIG wrapper code is generated from OCE's headers, so keeping things in sync is critical.

theliam, 2015-04-02:
I'm having the same problem. I don't quite understand what you did to get it working..
I have Swig 3.0.5 installed and am trying to install pythonocc 0.16.

What do you mean by 'changing the tag over'?

jmwright, 2015-04-28:
@theliam - It's been a long time, so I'm not sure if you're still interested in an answer on this, but I think he meant that you can clone a tagged version.

git clone --branch "0.16.0" https://github.com/tpaviot/pythonocc-core.git

jmwright, 2015-04-28:
Also, make sure to define the OCE include and library paths, as outlined in [this issue](https://github.com/tpaviot/pythonocc-core/issues/82).