tpaviot/pythonocc-core #91: make fails with missing files, while include paths are ignored

luminize, 2015-03-12:
Hi,

I'm trying to compile on debian jessie. I've succesfully built the oce library.
first I do the following like in install.md:
```shell
bas@mbp-jessie:~/pythonocc-core/build$ make clean
bas@mbp-jessie:~/pythonocc-core/build$ cmake ..
--  
-- Build 64bit
-- OpenCASCADE Community Edition found.
-- Platform:Unix
-- output_path: Unix/x86_64-MinSizeRel-64
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bas/pythonocc-core/build
bas@mbp-jessie:~/pythonocc-core/build$ make
[  0%] Swig source
Scanning dependencies of target _AIS
[  0%] Building CXX object CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o
/home/bas/pythonocc-core/build/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx:5505:39: fatal error: Graphic3d_ZLayerSettings.hxx: No such file or directory
 #include<Graphic3d_ZLayerSettings.hxx>
                                       ^
compilation terminated.
CMakeFiles/_AIS.dir/build.make:62: recipe for target 'CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o' failed
make[2]: *** [CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/_AIS.dir/all' failed
make[1]: *** [CMakeFiles/_AIS.dir/all] Error 2
Makefile:137: recipe for target 'all' failed
make: *** [all] Error 2
```

locating the file:
```shell
bas@mbp-jessie:~/pythonocc-core/build$ find /home/bas -name Graphic3d_ZLayer*.h* 
/home/bas/oce/src/Graphic3d/Graphic3d_ZLayerSettings.hxx
bas@mbp-jessie:~/pythonocc-core/build$ 
```

so I decide to use the arguments for the library and headers as per install.md info;
```shell
bas@mbp-jessie:~/pythonocc-core/build$ make clean
bas@mbp-jessie:~/pythonocc-core/build$ cmake .. -DOCE_INCLUDE_PATH=/home/bas/oce/inc -DOCE_LIB_PATH=/home/bas/oce/src
--  
-- Build 64bit
-- OpenCASCADE Community Edition found.
-- Platform:Unix
-- output_path: Unix/x86_64-MinSizeRel-64
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bas/pythonocc-core/build
bas@mbp-jessie:~/pythonocc-core/build$ 
```
but still the build fails with `make`.
```shell
bas@mbp-jessie:~/pythonocc-core/build$ make
[  0%] Swig source
Scanning dependencies of target _AIS
[  0%] Building CXX object CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o
/home/bas/pythonocc-core/build/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx:5505:39: fatal error: Graphic3d_ZLayerSettings.hxx: No such file or directory
 #include<Graphic3d_ZLayerSettings.hxx>
                                       ^
compilation terminated.
CMakeFiles/_AIS.dir/build.make:62: recipe for target 'CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o' failed
make[2]: *** [CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/_AIS.dir/all' failed
make[1]: *** [CMakeFiles/_AIS.dir/all] Error 2
Makefile:137: recipe for target 'all' failed
make: *** [all] Error 2
```
Could you tell me what I'm doing wrong? The location of the header file in `/home/bas/oce/src/Graphic3d` can't be found with the arguments I pass.
I'm probably missing something obvious.
Regards,
Bas

tpaviot, 2015-07-15:
Fixed by commit 03401ff

closed 2015-07-15 by tpaviot

ndiayefrancisco2000, 2015-11-09:
Makefile:137: recipe for target ‘all’ failed
make: *** [all] Error 2

how exactly was the problem fixed? I've got the same problem . Thanks in advance for the help.

ndiayefrancisco2000, 2015-11-09:
mine stopped at 13%