tpaviot/pythonocc-core #92: Incompatible with `swig 3.0.5`

scicalculator, 2015-03-13:
I initially compiled PythonOCC using the latest build of Swig 3.0.5. However, compilation using Swig3.0.5 builds PythonOCC in a manor that it fails the tests because of consistent out-of-scope calls to variables. 

As an example, in `Graphic3d.py` it uses `Quanitity_NOC_RED` without the variable being imported. Instead, the build imports `OCC.Quantity`, but then references quantity variables by their name as if they had been imported using `from OCC.Quantity import *`.

I fixed my installation by rebuilding with Swig 2.0.10.

jf---, 2015-05-20:
Hi @scicalculator , thanks this is useful info... 
Perhaps we should perform a swig compatibility check.

This might be a platform specific issue -- on osx, I dont have a problem with swig > 3.0.1.
Are you on linux?

mortbauer, 2015-06-16:
I had the same problem using swig 3.0.5 on Archlinux, i switched to swig 3.0.2 which works fine.

tpaviot, 2015-07-15:
There has been a change in swig 3.0.3 regarding the way default value are wrapped. The 'compactdefaultargs' feature must be enabled to follow the older behavior.

tpaviot, 2015-07-15:
Fixed by commits https://github.com/tpaviot/pythonocc-generator/commit/4dab39a54a90a6aa70548ed420becb990495dc92 and caa0c4f67bd632736f4335a21cc0438c2f9399ed. After that, pythonocc can be compiled with any swig version, 2 or 3.

closed 2015-07-15 by tpaviot