tpaviot/pythonocc-core #96: Aspect_DriverDefinitionError

HuntForRedOctober, 2015-04-24:
When running any of the examples I get the following error regardless of the backend I specify

C:\Users\xxxx\Downloads\pythonocc-core-0.16.0\examples>python core_geometry_airfoil.py
###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Traceback (most recent call last):
  File "core_geometry_airfoil.py", line 107, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display('wx')
  File "C:\Users\xxxx\Programs\pythonxy\python27\lib\site-packages\OCC\Display\SimpleGui.py", line 95, in init_display

    win.canva.InitDriver()
  File "C:\Users\xxxx\Programs\pythonxy\python27\lib\site-packages\OCC\Display\wxDisplay.py", line 108, in InitDriver
    self._display.Create()
  File "C:\Users\xxxx\Programs\pythonxy\python27\lib\site-packages\OCC\Display\OCCViewer.py", line 105, in Create
    self.Init(self._window_handle)
RuntimeError: Aspect_DriverDefinitionError
The specified module could not be found.

I'm using:
pythonOCC-0.16.0-win32-py27
Python 2.7.9 (default, Dec 10 2014, 12:24:55) [MSC v.1500 32 bit (Intel)] on win32
('Qt version:', '4.8.4')
('SIP version:', '4.14.2')
('PyQt version:', '4.9.6')
PySide 1.2.2
Wx 2.8

Any Idea?
I have used the pythonOCC-0.16.0-win32-py27.exe installer on a different machine to decompress and I have copied over the OCC folder to the sitepackages folder of my python installation. Could this be a problem?

tpaviot, 2015-04-26:
This is an issue with the OpenGL driver. But sure your graphic card driver is up-to-date. Which card model is it ?

HuntForRedOctober, 2015-04-27:
I have a nvidia Quadro 3000M, about 3-4 years old by now. The drivers should be reasonably up-to-date..
Thanks for your help
Stefano

<html><head></head><body><font color="black" face="arial" size="2">
This email and its attachments may contain confidential and/or privileged information.  If you have received them in error you must not use, copy or disclose their content to any person.  Please notify the sender immediately and then delete this email from your system.  This e-mail has been scanned for viruses, but it is the responsibility of the recipient to conduct their own security measures. Airbus Operations Limited is not liable for any loss or damage arising from the receipt or use of this e-mail. 

Airbus Operations Limited, a company registered in England and Wales, registration number, 3468788.  Registered office:  Pegasus House, Aerospace Avenue, Filton, Bristol, BS34 7PA, UK.
</font>
</body>
</html>

jf---, 2015-05-20:
Sounds like XQuartz / X11 is not running...
On which platform are you running, please state the pythonocc / OCE version...
Thanks

HuntForRedOctober, 2015-05-20:
I’m actually running on windows 7 using OCE 0.16.1 and python-occ 0.16.0

neerajs71, 2015-05-26:
I have the same problem on Windows-7. Running on Sony I5 Viao. RADEON Graphics driver.

neerajs71, 2015-05-26:
Is there any resolution to the issue? Any advice?

Rebbirobbe, 2015-05-29:
Hello, 

we got the same error on Windows 7 (64bit and 32bit).. 
(we tried to run the OCC-Box-example) 
but it seems that we have found the solution.
We had to set an environment variable for system:
CSF_GraphicShr=c:\py27\Lib\site-packages\OCC\libTKOpenGl.dll
(we use Python X,Y)
During the installation an environment variable is set, but only for the user who installed the software.

Greetings

tpaviot, 2015-05-29:
@Rebbirobbe That's the point actually, I forgot about that! @HuntForRedOctober can you please test that ? I may have to fix the installer script so that this environnement variable, on Windows, is set for all users.

HuntForRedOctober, 2015-06-01:
I tried with the environmental variable, but it didn’t work out..

But…

I copied all the dlls from the OCC folder in the site packages to the main python folder… and then it worked out OK..

tpaviot, 2015-06-01:
@HuntForRedOctober Ok, thanks for the tip. OCE dlls are installed to the site-packages/OCC directory, the same folder than the .py modules. It works on most machines, I didn't know it can be a problem.

closed 2015-06-20 by tpaviot