tpaviot/pythonocc-core #98: Travic-CI broken because of bad oce version

tpaviot, 2015-05-05:
Current travis-ci builds fail. pythonocc current master requires oce-0.16.x to compile, but the ppa from which oce binaries are pulled recently moved up to 0.17. Thus the compilation fails.

This can be solved using two solutions:

1. the freecad tems restore oce-0.16

2. we (at oce or pythonocc) provide a launchpad ppa for oce releases.

The second solution would be quite better.

jmwright, 2015-05-05:
I think an OCE PPA that's sync'd with the pythonocc PPA, and that we can be sure won't break when new versions are released (option 2), would be very helpful.

jmwright, 2015-05-05:
OCE 0.16 is available on [Alioth](http://anonscm.debian.org/cgit/debian-science/packages/oce.git). The [git-buildpackage](https://www.debian.org/doc/manuals/maint-guide/build.en.html) (section 6.5) utility is used to build Debian packages from git.

```
sudo apt-get install quilt libfreetype6-dev tcl8.5-dev tk8.5-dev libgl2ps-dev libfreeimage-dev
git clone https://anonscm.debian.org/git/debian-science/packages/oce.git
cd oce
sudo git-buildpackage --git-ignore-branch
```

In theory, it could just be a matter of placing the generated source packages in a PPA after that. I haven't tried it yet though because the build is taking quite awhile. I can report report back and keep pursuing this if it seems like an acceptable option.

tpaviot, 2015-05-06:
I didn't know about git-buildpackage, thank you for pointing this.

Launchpad has a farm build system, that prevents from compiling deb packages on your machine and upload them to launchpad. I already did it for pythonocc-core, I could do the same for oce, it's just that it's not straightforward and a good knowledge of debian packaging is required. It was quite a good think the freecad people take care of that.

closed 2015-05-07 by tpaviot