tpaviot/pythonocc-core #99: some examples fails

Jiloc, 2015-05-06:
Hello, i just installed pythonocc-core with miniconda following your instructions. Tests and most of the examples works fine, but a few of them crashes. Am I missing something or they were not updated with the last release?

- 1 

        ▶ python core_visualization_overpaint_viewer.py
        Traceback (most recent call last):
            File "core_visualization_overpaint_viewer.py", line 41, in <module>
              from OCC.Display.qtDisplay import qtViewer3d, get_qt_modules
        ImportError: No module named qtDisplay

I checked and I saw that I only have a file named pyqt4Display.py. 
qtDisplay.py is from your /src/addons/Display/ folder, and i wans't installed at all with conda!

- 2 

        ▶ python core_load_step_ap203_ocaf.py
        Traceback (most recent call last):
           File "core_load_step_ap203_ocaf.py", line 23, in <module>
             from OCC.TDocStd import Handle_TDocStd_Document
         ImportError: No module named TDocStd

This file does not exists at all in my installed lib.

- 3      

         ▶ python core_dimensions.py
          ###### 3D rendering pipe initialisation #####
         Display3d class initialization starting ...
         Aspect_DisplayConnection created.
         Graphic_Driver created.
         Cocoa window created.
         V3d_Viewer created.
         V3d_View created
         AIS_InteractiveContext created.
         Display3d class successfully initialized.
          ########################################
         unable to open /Users/jelleferinga/miniconda/envs/_build/share/oce-0.16/src/UnitsAPI/Lexi_Expr.dat for input
         unable to open /Users/jelleferinga/miniconda/envs/_build/share/oce-0.16/src/UnitsAPI/Units.dat for input
         Traceback (most recent call last):
           File "core_dimensions.py", line 35, in <module>
             display.Context.Display(rd.GetHandle())
         RuntimeError: Units_NoSuchType
         BAD LEXICON descriptor

NB: /Users/jelleferinga/ is a hardcoded path somewhere (it does not exist at all).

tpaviot, 2015-05-14:
@jf--- Can u please check what's wrong ?

jf---, 2015-05-20:
@Jiloc , this is most valuable feedback, many thanks for that...

so issue 3 relates to the OCE package, 1 and 2 are relating the pythonocc-core conda packages.
it will take me a bit to delve into this, but this feedback is *much* appreciated!
I'm on it...

-jelle

tpaviot, 2015-07-15:
@jf--- What's the status of this issue ?

jf---, 2015-12-27:
@tpaviot , this is resolved in the recent builds

closed 2016-01-22 by tpaviot