tpaviot/pythonocc-core #102: STANDARD_TYPE macro

Jiloc, 2015-05-12:
Hello, following the official tutorial ( http://www.opencascade.org/doc/occt-6.8.0/overview/html/occt__tutorial.html ) you find this line to check the type of faces:

    if(aSurface->DynamicType() == STANDARD_TYPE(Geom_Plane)){

In pythonocc i ended up using this condition to simulate the same condition:

    if surface.GetObject().DynamicType().GetObject().Name() == Geom_Plane.__name__:
     

Is there a simpler and more elegant way to do the same?

tpaviot, 2015-05-12:
I guess that the following should do the job:

    def isOCCInstance(instance, occ_class):
        return instance.GetObject().DynamicType().GetObject().Name() == occ_class.__name__
    if isOCCInstance(surface, Geom_Plane):
      blabla()

It's not simpler or more elegant, but this is more readable.

jmwright, 2015-05-12:
Is the face_is_plain() check in the attached example [here](https://mail.gna.org/public/pythonocc-users/2011-05/msg00013.html) still valid? That seems like a pretty clean way to do this.

jmwright, 2015-05-12:
The following seems to work in the script I've been writing for the example.
```python
from OCC.Geom import Geom_Plane, Handle_Geom_Plane

def face_is_plane(face):
    ''' Returns True if the TopoDS_Shape is a plane, False otherwise
    '''
    hs = BRep_Tool_Surface(face)
    downcast_result = Handle_Geom_Plane.DownCast(hs)
    # the handle is null if downcast failed or is not possible,
    # that is to say the face is not a plane
    if downcast_result.IsNull():
        return False
    else:
        return True
...
if face_is_plane(aFace):
    aPlane = Handle_Geom_Plane.DownCast(BRep_Tool_Surface(aFace)).GetObject()
...
```

Jiloc, 2015-05-12:
both are better then mine, thank you! @tpaviot just another question, do you think is it possible to port the macro from oce instead of using those workaround?

tpaviot, 2015-05-14:
The STANDARD_TYPE macro is just a ugly hack available from the Standard_Macro.hxx oce header:

    # define   STANDARD_TYPE(aType)   aType##_Type_()

I don't think it's worth working on wrapping something like that.

closed 2015-05-14 by tpaviot

Jiloc, 2015-05-14:
Thank you everything is clear!