tpaviot/pythonocc-core #104: classic bottle example missing

jf---, 2015-05-20:


jmwright, 2015-05-20:
I would be happy to submit my bottle example for review if it will help with this. I wrote it using pythonocc-core 0.16. It needs improvement, but it works and was adapted from older examples.

jf---, 2015-05-20:
Great, could you submit a PR?
Only comment is that I'd make casting the Geom_Plane from a face into a function.

Thanks @jmwright !

```python
myBody_traverser = Topo(myBody)
for aFace in myBody_traverser.faces():
    if face_is_plane(aFace):
        aPlane = Handle_Geom_Plane_downcast(BRep_Tool_surface(aFace)).GetObject()
        aPnt = aPlane.Location()
        aZ = aPnt.X()
        if aZ>zMax:
            zMax = aZ
            faceToRemove = aFace
```

```python

def geom_plane_from_face(aFace):
    """ returns the geometric plance entity from a planar surface"""
    return Handle_Geom_Plane_downcast(BRep_Tool_surface(aFace)).GetObject()

myBody_traverser = Topo(myBody)
for aFace in myBody_traverser.faces():
    if face_is_plane(aFace):
        aPlane = geom_plane_from_face(aFace)
        aPnt = aPlane.Location()
        aZ = aPnt.X()
        if aZ>zMax:
            zMax = aZ
            faceToRemove = aFace
```

jmwright, 2015-05-20:
@jf--- I made the change you suggested, but I am unable to find the equivalent of Topo in pythonocc-core. Therefore, I have to use a bulky while loop instead of iterating using a for loop. That's one of the things that I'm thinking can be improved.

jf---, 2015-05-22:
Cool! Code looks really clean & mint, thanks Jeremy...

@jmwright , OK, let me add the topology.py module, its a really key piece of functionality.
@tpaviot , what about src/addons/Utils/topology.py?

that PR would rework also the examples to use that module.
@jmwright , are you OK to wait for this PR to be merged post merging topology.py?

jmwright, 2015-05-22:
@jf--- Thanks. Glad the code looks good. I'm ok waiting for my PR to be merged. I had actually expected it to take awhile while the example was improved (with things like the inclusion of topology.py).

tpaviot, 2015-05-25:
Fixed by PR #105

closed 2015-05-25 by tpaviot