tpaviot/pythonocc-core #106: No matching distribution found for pythonocc-core when using pip install

ldgarcia, 2015-05-24:
I tried installing pythonocc-core with `pip`, but an error occurs:

````bash
$ pip search pythonocc
pythonocc-core     - pythonocc-core wraps the OpenCASCADE CAD Kernel
$ pip install pythonocc-core
Collecting pythonocc-core
  Could not find a version that satisfies the requirement pythonocc-core (from versions: )
No matching distribution found for pythonocc-core
````

jf---, 2015-05-24:
good point @ldgarcia ...
I think its important that pythonocc is present in PyPI since its such an authorative reference for the python eco system. However, the right approach would be that trying to install by pip would result in getting a link to instructions explaining you what your options are on the specific platform...

thanks again,

-jelle

ldgarcia, 2015-05-30:
Yes, that would be a very good approach @jf---

tpaviot, 2016-02-18:
pip support is not planned.

closed 2016-02-18 by tpaviot