tpaviot/pythonocc-core #107: install on Ubuntu 14.04.2 via ppm fails

rbakker, 2015-06-09:
Installation:
sudo apt-get install pythonocc-core
Note that this installs pythonocc 0.16 with oce-draw 0.17

The message is:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/OCC/BRepPrimAPI.py", line 32, in <module>
    _BRepPrimAPI = swig_import_helper()
  File "/usr/lib/python2.7/dist-packages/OCC/BRepPrimAPI.py", line 28, in swig_import_helper
    _mod = imp.load_module('_BRepPrimAPI', fp, pathname, description)
ImportError: libTKernel.so.9: cannot open shared object file: No such file or directory

I checked the LD_Library_Path, it includes 
/usr/lib/x86_64-linux-gnu/
which contains  libTKernel.so.10
so this seems to be a versioning issue. Do I have to recompile?

tpaviot, 2015-06-09:
This is an issue with the free-cad mainteners repository. They have upgraded oce to the latest version (0.17), 0.16 is unfortunately not provided anymore.This breaks apt-get pythonocc installation, I have to upload an oce-0.16 to my own launchpad repository and update the ppa url.

rbakker, 2015-06-09:
Thanks for the update. I managed to install oce + pythonocc, it is a lot of work with the versioning issues, this would be a lot easiers if managed by the ppa url.
Now let me discover the wonders of pythonocc!

closed 2015-06-09 by rbakker

jmwright, 2015-06-10:
It's not official or anything, but this is a script that I use to set up my PythonOCC dev environments. https://github.com/jmwright/pythonocc_oce_setup