tpaviot/pythonocc-core #108: Hello world example: TypeError: int() argument must be a string or a number, not 'PyCObject'

doczoidberg, 2015-06-12:
I tried to run the Hello World example but get the error:

```
Traceback (most recent call last):
  File "C:\Program Files (x86)\JetBrains\PyCharm Community Edition 4.5.1\helpers\pydev\pydevd.py", line 2357, in <module>
    globals = debugger.run(setup['file'], None, None, is_module)
  File "C:\Program Files (x86)\JetBrains\PyCharm Community Edition 4.5.1\helpers\pydev\pydevd.py", line 1777, in run
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "D:/EO/freecadscripts4/x3domtests.py", line 6, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display()
  File "C:\Python27\Lib\site-packages\OCC\Display\SimpleGui.py", line 164, in init_display
    win.canva.InitDriver()
  File "C:\Python27\Lib\site-packages\OCC\Display\pysideDisplay.py", line 79, in InitDriver
    self._display = OCCViewer.Viewer3d(self.GetHandle())
  File "C:\Python27\Lib\site-packages\OCC\Display\pysideDisplay.py", line 55, in GetHandle
    return int(self.winId())
TypeError: int() argument must be a string or a number, not 'PyCObject'
```

jf---, 2015-06-12:
might be a duplicate with issue #96

doczoidberg, 2015-06-12:
I use PySide by the way. I couldn't install PyQt4 or wxPython

jf---, 2015-06-12:
That's not the issue here, but good to know...

tpaviot, 2015-06-12:
This seems related to a difference between pyside and PyQt, see http://srinikom.github.io/pyside-bz-archive/523.html

doczoidberg, 2015-06-12:
I installed wxPython via the setup und now it works.

tpaviot, 2015-06-12:
I reopen the issue, it has to be fixed so that it's possible to use PySide under Win.

closed 2015-06-20 by tpaviot