tpaviot/pythonocc-core #115: Compiling Win64 bit Python27 version of Pythonocc

chenkianwee, 2015-06-29:
I need a win64 python27 pythonocc. I followed the instructions and try to compile a 64bit version, however I am receiving this error. 

[image]

Has anyone had this problem before while trying to compile a 64bit version ?

tpaviot, 2015-07-01:
64 or 32 bit should not be the problem. I sue to compile pythonocc using mingw32 on windows, not msvc. According to the screenshot, it seems TKernel library is not found. Did you download oce-0.16 windows precompiled binaries ? can you please rather test to compile in Release mode ?

chenkianwee, 2015-07-03:
Thanks for the reply. I compiled my own oce-0.16, let me download the precompiled binaries and try it out with mingw32, lets hope I will be able to succeed this time.

chenkianwee, 2015-07-03:
Hi, 
I again attempt to build the 64bit version. First of all this time I downloaded the precompiled binaries of oce 0.16.0
1.) So I tried to build the project with msvc, considering with the new oce precompiled binaries it will be able to find all the necessary files this time. But it failed. 
[image]

2.) So I decided to build it with mingw, so I cmake pythonocc core for mingw compiler.
[image]

3.) It manages to generate the files with an error of "error making directory src/vizualisation"
4.) From here i try to compile the files with make command from the msys.bat file, but nothing happen. 
[image]

Am I doing something wrong here? Is there a guide I can look at to compile the pythonocc? Sorry to be so long-winded

tpaviot, 2015-07-31:
Mingw64 copilation fixes has been released with pythonocc-0.16.2. Please try again to compile it on your own, or download prempiled binaries for win64/python27 from the download page.

closed 2015-07-31 by tpaviot