tpaviot/pythonocc-core #122: Error: gp_Pnt is not defined

AmitBedaka, 2015-07-28:
I am trying to run the Bottle.py code in OCC 6.9 in Ubuntu to understand the PythonOCC. Its more challenging to understand the class and it uses. Can anyone please suggest me the possible way to understand pythonocc. It will be great help for me. Kindly give ur valuable time to reply on email: akbedaka@gmail.com.

jf---, 2015-07-28:
Traceback please?

tpaviot, 2015-07-30:
@AmitBedaka Invalid issue report. Please report a reduced scope question/issue with precise description.

AmitBedaka, 2015-07-30:
Thank you for your kind reply. I need to do my research using PythonOCC. I
am trying to solve bottle.py example but I am not able to execute it using
PythonOCC in Ubuntu. Kindly give your valuable time and suggestions.

Thanking you,

Kind regards,

*Amit Kumar Bedaka*
*M.Des (Mechanical System Design)*

*Indian Institute of Information Technology Design and Manufacturing,
Kancheepuram, (Institute of National Importance / Center of Excellence,
MHRD Govt of India)Chennai - 600 127, *
*Tamil Nadu, INDIA.*

AmitBedaka, 2015-08-24:
Can anyone help me to sort out the above issues?

tpaviot, 2015-08-24:
OCC 6.9 is not supported by pythonocc-0.16.x. You *must* download/install oce-0.16 or oce-0.16.1

AmitBedaka, 2015-08-24:
Thank you for your kind reply.
ᐧ

*Amit Kumar Bedaka*
*M.Des (Mechanical System Design)*

*Indian Institute of Information Technology Design and Manufacturing,
Kancheepuram, (Institute of National Importance / Center of Excellence,
MHRD Govt of India)Chennai - 600 127, *
*Tamil Nadu, INDIA.*

tpaviot, 2015-08-24:
Download here: https://github.com/tpaviot/oce/releases/tag/OCE-0.16.1

closed 2015-08-24 by tpaviot