tpaviot/pythonocc-core #124: BRepOffsetAPI_MakeFilling sefgaults

jf---, 2015-07-28:
see [this commit](https://github.com/jf---/pythonocc-core/commit/597ae92f416ef1b18d17e9bb894b40e8cdfc2a9b), port of a cool example of [an earlier thread](https://groups.google.com/forum/#!topic/pythonocc/4DkO1R67m0k) on the pyocc mailing list

jf---, 2015-07-28:
Strange, looked at the underlying changes in OCE, but there are [basically none](https://github.com/tpaviot/oce/blame/master/src/BRepFill/BRepFill_Filling.cxx). Could this be an issue with objects going out of scope / getting swepped up by the GC or so?

jf---, 2015-07-28:
no difference with:
    from OCC.BRepFill import BRepFill_Filling
    brep_plate_builder = BRepFill_Filling()

tpaviot, 2015-07-30:
@jf--- Can you please be more explicit about this issue.

jf---, 2015-07-30:
Sure, so I manage to run this example on the last pythonocc version, but can't get it to run on pythonocc-core 0.16.

tpaviot, 2016-04-27:
Fixed by commit 5dda4c3
[image]

closed 2016-04-27 by tpaviot

jf---, 2016-04-27:
w000000h00000 ! Excited to see this Thomas... I think its one of the harder examples I wrote...
Thomas, I've been thinking about writing some pyocc blog posts; this might make an interesting candidate; the continuity between the surfaces yields what is known as a [Class A surface](http://english.avrotas.com/class-a-surface-modeling/) that is curvature continuity from one patch to the other. This is traditionally the domain of Catia ( with the ICE plugin ) and Alias and sorts. So, to be able to generate this kind of surface type through code / pyocc is kind of something noteworthy.

tpaviot, 2016-04-28:
This example passes on my windows machine, on Linux too, but seems to fail on osx according to travis (https://travis-ci.org/tpaviot/pythonocc-core/jobs/126295771)

jf---, 2016-04-28:
aha... could this have anything to do with [Workaround clang optimizations for null references #604](https://github.com/tpaviot/oce/pull/604)

tpaviot, 2016-04-28:
maybe. This shoulb be investigated with valgrind. I notify @rainman110 , for information

rainman110, 2016-04-28:
@tpaviot @jf--- I'll gonna have a look into it today or tomorrow. Lets see, what valgrind reports. There might be some compiler specific issues of course, but I don't think that this is related to the workaround.

jf---, 2016-04-28:
That's awesome @rainman110 , thanks... 
Sorry, I didnt phrase my argument well -- I meant that perhaps the test on OSX doesnt complete for _lack_ of your fix ;) that is I'm not sure whether travis picks up OCE with your patches.

rainman110, 2016-04-28:
@jf--- Travis should already include the clang specific patches, if I remember correctly ;) At least, it's inside the recipe. Else, many other tests would fail too.

jf---, 2016-04-28:
duhh... thanks for the reminder @rainman110 ... sorry, that's pretty obvious...