tpaviot/pythonocc-core #125: compilation error:  AIS_AngleDimension.hxx: No such file or directory

liuhuanjim013, 2015-07-28:
I was not able to compile pythonocc-core 0.16.2 using swig-2.0.10 and oce-0.16 (with -DOCE_DISABLE_X11:BOOL=ON). 

Once I removed the OCE_DISABLE_X11 flag and recompiled oce, pythonocc could be compiled.

Since OCE_DISABLE_X11 is not a documented flag, it is understandably not supported. Maybe a better error message or adding a build option in pythonocc would be better. Or if it is a required header file, include it even when OCE_DISABLE_X11 is on?

jf---, 2015-07-29:
On which platform are you?
( I'm guessing Linux )

IIRC, the OCE_DISABLE_X11 came about when integrating the osx Cocoa backend.
@tpaviot will have the authorative answer, and yes, perhaps we can do a better job documentation the flag.

liuhuanjim013, 2015-07-29:
Thanks for the comments! I'm on Debian Wheezy.

jf---, 2015-07-29:
OK, thats helps. So yeah, disabling X11 is not possible for linux, so that's here for compiling on other platforms like OSX that do not rely on X11 as a GUI framework.

liuhuanjim013, 2015-07-29:
The problem led to me disabling X11 is due to my custom graphics set up. I was having this linking error

```
Linking CXX executable OCAFExport_test
/usr/lib/gcc/x86_64-linux-gnu/4.7/../../../x86_64-linux-gnu/libGL.so: undefined reference to `_glapi_tls_Dispatch'
```

My solution in the end was to disable building tests by adding  -DOCE_TESTING:BOOL=OFF

jf---, 2015-07-30:
Ah, that's an interesting find @liuhuanjim013 
Can I ask you to close this issue, and make another one for the linking error?
Please mention that your on Debian Wheezy, that's pretty relevant info...
thanks...

closed 2015-07-30 by liuhuanjim013

liuhuanjim013, 2015-07-30:
I don't think this is oce's fault. I have a pretty non-standard (in other words, “broken") graphics set up and OCAFExport_test is not doing anything wrong, it's simply linking to the system's default libGL.so, which incorrectly points to the wrong library (which i know how to fix).

My point of opening this issue was to start the discussion / understand the reasoning behind the exclusion of important header files when certain flags are on. I guess the take away is that DISABLE_X11 is not supported.

Thanks!

jf---, 2015-07-30:
Alright, interesting to know & good to hear you consider this not a OCE issue... :sunglasses: