tpaviot/pythonocc-core #126: test fails the first time but succeeds the second time

liuhuanjim013, 2015-07-29:
output of the first time python run_tests.py was run:

```
...
test_read_step_file (core_ocaf_unittest.TestOCAF)
Reads the previous step file ... FAIL
test_write_step_file (core_ocaf_unittest.TestOCAF)
Exports a colored box into a STEP file ... 
...
FAIL: test_read_step_file (core_ocaf_unittest.TestOCAF)
Reads the previous step file
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/liuhuan/src/pythonocc-core/test/core_ocaf_unittest.py", line 107, in test_read_step_file
    assert(labels.Length() == 1)
AssertionError

----------------------------------------------------------------------
Ran 46 tests in 0.171s

FAILED (failures=1)
```

second time
```
Ran 46 tests in 0.163s

OK
```

it seems that the read test was executed before the write test unexpectedly. instead of relying on the execution order of the unit test framework, maybe merging the two into one unit test? so that tests can be executed in parallel.

jf---, 2015-07-29:
interesting, thanks for the heads up... 
still, that's a pretty fast test suite :sunglasses:

liuhuanjim013, 2015-07-29:
haha yeah ssd ftw. also the test artifacts stay around after the tests finish, maybe there should be a clean-up step as well to ensure that tests start with the same initial condition.

tpaviot, 2015-07-30:
@liuhuanjim013 The fix is in the commit 9dc5465c7fbf3a90308dbd58fd593701b4816402

closed 2015-07-31 by tpaviot

liuhuanjim013, 2015-07-31:
cool! was not able to pass all the tests, but the issue was addressed.

tpaviot, 2015-07-31:
@liuhuanjim013 Please open new tickets on the issue tracker for the tests that don't pass on your machine. Don't forget to tell us about your machine/os/pythonocc version you've been working with.