tpaviot/pythonocc-core #128: Installation problem

bs7, 2015-07-31:
Hello,

I successfully built oce but when it comes to build pythonocc-core it stop with the error below.
I also tried the "conda" strategy but the package was missing.

+------------------------------------------------------------------------------------------------------------------------+
root@PC-BS-NUC:/usr/local/src/pythonocc-core/cmake-build# cmake ..
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
--  
-- Found PythonInterp: /root/miniconda/bin/python (found version "2.7.8") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found version "2.7.6") 
-- Build 64bit
-- Found SWIG: /usr/bin/swig2.0 (found version "2.0.11") 
-- OpenCASCADE Community Edition (OCE) found.
-- Platform:Unix
-- output_path: Unix/x86_64-MinSizeRel-64
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/local/src/pythonocc-core/cmake-build
root@PC-BS-NUC:/usr/local/src/pythonocc-core/cmake-build# make
[  0%] Swig source
Scanning dependencies of target _AIS
[  0%] Building CXX object CMakeFiles/_AIS.dir/src/SWIG_files/wrapper/AISPYTHON_wrap.cxx.o
/usr/local/src/pythonocc-core/cmake-build/src/SWIG_files/wrapper/AISPYTHON_wrap.cxx:4600:33: fatal error: AIS_ConnectedShape.hxx: No such file or directory
 #include<AIS_ConnectedShape.hxx>
+------------------------------------------------------------------------------------------------------------------------+

tpaviot, 2015-07-31:
@bs7 It's usually because the oce include directory is not right. Can you please relaunch the build and copy/paste the trace here with:
<pre>
make VERBOSE=1
</pre>

@jf--- What about the conda part of the issue ?

jf---, 2015-07-31:
@tpaviot  , at the moment, a package only exists for osx, where @bs7 runs linux

bs7, 2015-08-03:
@tpaviot: it was a version mismatch. Maybe a short hint in the INSTALL.md just to underline that oce and pythonocc must have same version could may be helpful for newcomer.

@jf--- yes it is clearly stated that it is only for osx, I wish it could come for Linux too.

closed 2015-08-03 by bs7