tpaviot/pythonocc-core #129: Update PPA for Ubuntu 15.04 ?

stuaxo, 2015-07-31:
Hi,
   Is there any chance of updating the PPA for Vivid (15.04) ?

S

tpaviot, 2015-07-31:
I didn't work on the ppa part for a while, it's on my todo list.

jf---, 2016-02-17:
@tpaviot , shall we point to conda rather than offering distro specific packages?

stuaxo, 2016-02-17:
If you are doing that, would it be possible to make a version installable from pip ?

jf---, 2016-02-17:
hi @stuaxo , nope, that's beyond the technical scope of pip...
we support binary installation ( pythonocc + dependencies ) [through conda](https://anaconda.org/DLR-SC). there's only so much you can do in terms of installing dependencies in pip, hence conda.

tpaviot, 2016-04-27:
Please use conda.

closed 2016-04-27 by tpaviot

stuaxo, 2016-04-27:
Cheers, unfortunately every time I've tried conda it doesn't have the things in it I like to use.

If I play try pythonocc in the future I might have a look at getting pythonocc working with vext (a horrible hack I've made to make things that don't work in virtualenv accessible from virtualenv).