tpaviot/pythonocc-core #136: PYOCC ON LINUX ON PYTHONANYWHERE

pyenthu, 2015-08-03:
Is there any convenient way to install PYOCC on PYTHONANYWHERE? Sudo and get is not allowed. 
 I have a working version on my windows laptop but installation on Linux on the cloud is required to create a webapp.

jf---, 2015-08-03:
Interesting idea, what OS is pythonanywhere based on?

pyenthu, 2015-08-04:
It is Linux-64. I did try to compile it under my home directory. Managed to compile the oce using cmake and make, following the instructions on google groups. I was particular about the tag for 16.0.1 oce. However when I tried to compile pyocc-core it had an error "AIS_ConnectedShape.hxx" not found. This should have been there. So it seems that somehow the tag for the older version of OCE is not working. I am not sure how to resolve the issue. Not an expert on this but trying to get it fixed based on the post below. Is there a clean version of oce 16.0.1 somewhere I could start with?
https://groups.google.com/forum/#!topic/pythonocc/uzMv3i5ccv0

Having explored various graphics packages as a recently acquired hobby I think that having occ on the backend and parsing the objects to threejs with your tesselator is very promising if I can get it installed.

tpaviot, 2015-08-25:
@pyenthu You should manually pass paths where you installed oce. Did you ``` make install ``` for oce ?

pyenthu, 2015-09-18:
Sorry for the late reply. I managed to install it. Had stopped working on
it for a while but will see if I can actually get it to work.

tpaviot, 2016-01-22:
@pyenthu Any progress related to this issue ?

pyenthu, 2016-01-22:
Yes it is working now. :). I somehow got it working. It was a few months ago. I would request a pip install if possible sometime.

closed 2016-01-22 by tpaviot