tpaviot/pythonocc-core #138: Display initialization failure with win32/py34 and PyQt

tpaviot, 2015-08-12:
Trying to run the core_helloworld.py on Windows 32 bit/Python34/pythonOCC-0.16.2/ produces the following issue:
<pre>
Traceback (most recent call last):
  File "core_helloworld.py", line 21, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display()
  File "c:\Python34\lib\site-packages\OCC\Display\SimpleGui.py", line 159, in in
it_display
    win.canva.InitDriver()
  File "c:\Python34\lib\site-packages\OCC\Display\qtDisplay.py", line 106, in In
itDriver
    self._display.Create()
  File "c:\Python34\lib\site-packages\OCC\Display\OCCViewer.py", line 119, in Cr
eate
    self.Init(self._window_handle)
TypeError: in method 'Display3d_Init', argument 2 of type 'long'
</pre>

tpaviot, 2015-08-12:
Actually, PyQt4 winId() QWidget method can returns a sip.voitptr instead of an int.

tpaviot, 2015-08-12:
Commit 36cf660 fixes the issue: the sip.voiptr is converted to an integer before being passed to the Init method

tpaviot, 2015-08-12:
Commit f91cd0e546c403dacf34c64089ce9a0816c5df51 created a regression

tpaviot, 2015-08-12:
Fixed by PR #140

closed 2015-08-12 by tpaviot

chenkianwee, 2015-12-15:
Hi I am facing the same problem using PythonOCC win64/py27. This is the error message i get when I run the example.

Using PyQt4

Traceback (most recent call last):
  File "C:\Users\smrckwe\Desktop\occtest.py", line 4, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display()
  File "C:\Python27\lib\site-packages\OCC\Display\SimpleGui.py", line 159, in init_display
    win.canva.InitDriver()
  File "C:\Python27\lib\site-packages\OCC\Display\qtDisplay.py", line 106, in InitDriver
    self._display.Create()
  File "C:\Python27\lib\site-packages\OCC\Display\OCCViewer.py", line 118, in Create
    self.Init(self._window_handle)
TypeError: in method 'Display3d_Init', argument 2 of type 'long'

jf---, 2015-12-18:
hi @chenkianwee ,

This is a known and old problem, and already fixed...
So, by updating your installation you will be able to resolve this issue...

chenkianwee, 2015-12-21:
Oh I see the pythonOCC-0.16.2-win64-py27.exe I downloaded is a july release. Thanks for the reply