tpaviot/pythonocc-core #139: Messages rendering on linux/pyqt4

tpaviot, 2015-08-12:
Messages are not correctly rendered on linux/pyqt4. Below screenshot of the core_topology_glue.py example.

[image]

jf---, 2015-08-12:
yikes... 
do you suspect that is this an issue upstream, I mean with OCE rather than pythonocc-core?

tpaviot, 2015-08-12:
Hard to say, it also may come from my graphic driver

jf---, 2015-08-12:
are you running in vmware Thomas?

tpaviot, 2015-08-13:
no, a dual boot Ubuntu/Win32 on an Asus notebook

jf---, 2016-02-17:
@tpaviot ,

same here...

actually, slightly worse: in order to have messages displayed on OSX, the font has to be set explicitly.
the alternative of supplying a ``Handle_Prs3d_Drawer`` argument fails ( no messages shown )
only explicitly setting a path to a font pans out.

could this be a version issue? OCC 6.8.0 release notes state:

- FreeType 2.4.11-2.5.3 
- FreeImage 3.16.0

I'm on freeimage 3.17 for instance...

```python
    def DisplayMessage(self, point, text_to_write, height=None, message_color=None, update=False):
        """
        :point: a gp_Pnt or gp_Pnt2d instance
        :text_to_write: a string
        :message_color: triple with the range 0-1
        """
        aPresentation = Prs3d_Presentation(self._struc_mgr)
        drawer = Prs3d_Drawer()

        text_aspect = Prs3d_TextAspect()
        text_aspect.SetFont("/System/Library/Fonts/HelveticaNeue.dfont")
        text_aspect.SetSpace(1)

        text_aspect.SetHeightWidthRatio(0.5)
```

[image]

jf---, 2016-02-17:
tested with freeimage 3.16.0 rather than 3.17.0, results in same issue

jf---, 2016-09-26:
@tpaviot , @rainman110 , could we prioritize this issue?
I realize this is mostly OCE related...

I'm on osx, and this issue is still outstanding ( using the latest DLR-SC binaries, thanks @rainman110  )

[image]

rainman110, 2016-09-26:
@jf--- @tpaviot I my case, I can't even render fonts. The x,y,z coordinate axes labels are invisible on OS X.

Unfortunetely, I have no clue, what the reason for this issue is.

jf---, 2016-09-26:
hi @rainman110 thanks for your feedback, yes same thing here, since using the same binaries...

is it possible that we're using a newer Freetype lib, eg 3.17 rather than 3.16?

rainman110, 2016-09-26:
@jf---  I am using the freetype library, that ships with conda. This is version 2.55 at the moment.

Freeimage is not used for font rendering, as far as I know.

@jf--- On my Hackintosh (El Capitan), I never saw any fonts in OCE. I thought I would have been a problem, because it is a Hackintosh and on my older MacBook (10.9), the fonts worked.

It seems, that the issue also depends on the version of OS X. Still, the issue exists probably on all platforms, but is mainly visible on Mac (and obviously for some on Linuxes as well).

rainman110, 2016-09-26:
@jf--- Do you have a minimal but complete example for the issue ( https://github.com/tpaviot/pythonocc-core/issues/139#issuecomment-185177170 seems interesting)? I am thinking of an example with only a text rendering, no geometries etc...

jf---, 2016-09-26:
@rainman110 sure, good idea...

here the calls for text rendering are passed through a function rather than calling `display.DisplayMessage`, such that all calls for text rendering are self-contained:

```python
from OCC.Display.OCCViewer import color, to_string
from OCC.Display.SimpleGui import init_display
from OCC.Prs3d import Prs3d_Presentation, Prs3d_TextAspect, Prs3d_Text
from OCC.gp import gp_Pnt, gp_Pnt2d

display, start_display, add_menu, add_function_to_menu = init_display()

def DisplayMessage(point, text_to_write, height=None, message_color=None, update=False):
    """
    :point: a gp_Pnt or gp_Pnt2d instance
    :text_to_write: a string
    :message_color: triple with the range 0-1
    """
    aPresentation = Prs3d_Presentation(display._struc_mgr)
    text_aspect = Prs3d_TextAspect()

    if message_color is not None:
        text_aspect.SetColor(color(*message_color))
    if height is not None:
        text_aspect.SetHeight(height)
    if isinstance(point, gp_Pnt2d):
        point = gp_Pnt(point.X(), point.Y(), 0)
    Prs3d_Text.Draw(aPresentation.GetHandle(),
                    text_aspect.GetHandle(),
                    to_string(text_to_write),
                    point)
    aPresentation.Display()
    # @TODO: it would be more coherent if a AIS_InteractiveObject
    # is be returned
    if update:
        display.Repaint()
    return aPresentation

DisplayMessage(gp_Pnt(), "text rendering problems")

display.FitAll()
start_display()
```

jf---, 2016-09-26:
looks like OCE 0.17 works with [FreeType 2.53](https://www.opencascade.com/doc/occt-6.8.0/overview/html/index.html#OCCT_OVW_SECTION_5)

[image]

rainman110, 2016-09-26:
@jf--- In this example, I get the same stripes as in the first post for some of the letters on **windows**, **qt5**.

But the text at the trihedron is okay.

[image]

rainman110, 2016-09-26:
@jf--- Same issue with freetype 2.5.3. I just tested it.

rainman110, 2016-09-26:
@jf---,  @tpaviot  I think we need a mininal oce example for this. With python in between, debugging is too complex.

jf---, 2016-09-26:
@rainman110 , good to know `freetype` isnt the culprit... alas I can't contribute much for a C++ prototype, but I see how this makes sense...

tpaviot, 2016-09-27:
@jf--- No time to work on this issue unfortunately. I can reproduce the issue on my windows machine, not tested on Linux. Quite sure it's not OCE related, we should have received feedback fro mOCE users. I guess it's the way text is rendered in pythonocc, I suspect the DisplayMessage method to be the cause. No sure though.

jf---, 2016-09-27:
Lets accept that OCE isnt the causing the issue here.

Now, here's something interesting:

```python
# text_aspect.SetFont("/System/Library/Fonts/HelveticaNeue.dfont")
text_aspect.SetFont("/Users/jelleferinga/Library/Fonts/aaux_pro_bd_osf.ttf")
```

- if i comment out calls to `text_aspect.SetFont`, no text shows up
- If I call `text_aspect.SetFont`, it has no impact on the font rendered in the GUI

rainman110, 2016-09-27:
@jf--- @tpaviot  I definetly have the issue with the missing fonts at the trihedron also with our oce-based app.

You are both using OS X, right? Could you please download https://github.com/DLR-SC/tigl/releases/download/v2.1.7/TIGL-2.1.7-Darwin.dmg, open the TiGLViewer App and tell me, whether your trihedron axes fonts are visible or not. In my case, it work with my MacBook Pro, but my Hackintosh does not show the fonts.

jf---, 2016-09-27:
hi @rainman110 , let me check... 
IIRC, @tpaviot is on windows again these days

jf---, 2016-09-27:
[image]

rainman110, 2016-09-27:
Okay... this looks good. Strange though, that it doesn't work an my hackintosh.

jf---, 2016-09-27:
here's a minimal example using `Graphic3d`, which is what the OCC guide tells you to do.
still, same problem...

```
from __future__ import print_function

import string

from OCC.Display.SimpleGui import init_display
from OCC.Graphic3d import Graphic3d_Vertex
from OCC.Prs3d import Prs3d_Root_CurrentGroup, Prs3d_Presentation

display, start_display, add_menu, add_function_to_menu = init_display()

def create_ogl_group(display):
    """
    create a group that will store an OpenGL buffer
    """
    aPresentation = Prs3d_Presentation(display._struc_mgr)
    group = Prs3d_Root_CurrentGroup(aPresentation.GetHandle()).GetObject()
    return aPresentation, group

def draw_txt(display):
    a_presentation, group = create_ogl_group(display)
    v = Graphic3d_Vertex(1, 1, 1)
    group.Text(string.digits, v, 16, False)
    a_presentation.Display()

def main():
    draw_txt(display)
    start_display()

if __name__ == "__main__":
    main()
```

lalvarezt, 2016-11-09:
@tpaviot @jf--- With the latest OCE library (6.9.1) this issue is fixed. I tested all of the examples that you proposed. Here are some pics

rainman110, 2016-11-10:
@lalvarezt how did you test this? As Far as I know, pythonocc 0.17 suppports only OpenCASCADE 6.8.0. 6.9.1 should be binary incompatible .

tpaviot, 2016-11-10:
@rainman110 he generated the wrappers for current oce master (occt691)

lalvarezt, 2016-11-10:
@rainman110 I'm using pyqt5, but that's not the issue, I had the same problems when I was using the 6.8.0 version.

tpaviot, 2017-04-28:
@rainman110 @lalvarezt @jf--- This is fixed in current master supportin oce-0.18 (occt691)

[image]

closed 2017-04-28 by tpaviot

rainman110, 2017-07-04:
I know this is an old issue, but I finally found the root of the issue:
https://tracker.dev.opencascade.org/view.php?id=25691

OCE 0.17 can only be used with freetype 2.5.3 or older, else it will trigger this issue.