tpaviot/pythonocc-core #143: Compile Error in PythonOCC -- Anaconda Python compile with python OCC

manugoswami, 2015-08-19:
Unable to compile python OCC. Specs below
 
OS                                          :               Windows XP
Python OCC download :               pythonocc-core-0.16.0
OCC download                  :               OCE-0.16.1-Mingw32
Python                                 :               Anaconda python 2.7
For cmake                           :                cmake -G"MinGW Makefiles" -DOCE_INCLUDE_PATH=D:\Manu\OCE-0.16.1-Mingw32\OCE-0.16.1-Mingw32\include\oce –
                                                                DOCE_LIB_PATH=D:\Manu\OCE-0.16.1-Mingw32\OCE-0.16.1-Mingw32\Win32\lib ..
Errors                                    :               on issuing command mingw32-make      multiple errors like undefined reference to `_imp___Py_NoneStruct'

tpaviot, 2015-08-19:
@jf--- can you please have a lookt at this issue ?

manugoswami, 2015-08-21:
Hello Guys !  got the answer from website http://eli.thegreenplace.net/2008/06/28/compiling-python-extensions-with-distutils-and-mingw/   
the commands are
pexports.exe python27.dll > python27.def
dlltool --dllname python27.dll --def python27.def --output-lib libpython27.a
and placed the libpython27.a in libs directory of conda distribution.

then to make used the command 
mingw32-make -I D:\Manu\OCE-0.16.1-Mingw32\OCE-0.16.1-Mingw32\include\oce -I D:\Manu\OCE-0.16.1-Mingw32\OCE-0.16.1-Mingw32\Win32\lib  -i

mingw32-make install

manugoswami, 2015-08-21:
Please include these in the necessary install file

manugoswami, 2015-08-21:
The examples from website http://jelleferinga.com/files/pythonOCC_SVN/website/ are working fine

>>> from OCC.gp import *
>>> from OCC.GCE2d import *
>>> P1 = gp_Pnt2d(1,2)
>>> P2 = gp_Pnt2d(3,-5.1)
>>> makeline = GCE2d_MakeLine(P1,P2)
>>> makeline.IsDone()
1
>>> line = makeline.Value()

================= and ==========================
>>> from OCC.gp import *
>>> vec1 = gp_Vec(1,2,3)
>>> vec2 = gp_Vec(-1,-2,-3)
>>> vec1.Magnitude()
3.7416573867739413
>>> vec2.Magnitude()
3.7416573867739413
>> vec3 = vec1 + vec2
>> vec3.Magnitude()
0.0
=======================================  

Now the issue is that your simple example

from OCC.Display.SimpleGui import init_display                         --> works fine
from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox             --> works fine 
 
display, start_display, add_menu, add_function_to_menu = init_display()  --> gives error Quantity_NOC_Red is not defined

my_box = BRepPrimAPI_MakeBox(10., 20., 30.).Shape()
 
display.DisplayShape(my_box, update=True)
start_display()

manugoswami, 2015-08-21:
Any way  thanks for a good piece of software ! (but please answer my query also ;)

jf---, 2015-08-21:
@manugoswami , that's great... thanks for sharing the results.
so, the conda bindary is osx only for the moment, that's why you probably had a hard time to get it going.
we'll integrate your feedback such to provide more platforms with the next release...

jf---, 2016-02-17:
@tpaviot , lets close this?

closed 2016-02-18 by tpaviot