tpaviot/pythonocc-core #144: How-to compile Python-OCC with OCE 0.16 src

Valber, 2015-08-25:
I Download OCE 0.16.1 https://github.com/tpaviot/oce/releases/tag/OCE-0.16.1

In cmake-gui activate

```
OCE_ADD_HEADERS
OCE_COPY_HEADERS_BUILD
OCE_DRAW
```

oce compiled OK

clone pythonocc-core from git
```
git clone https://github.com/tpaviot/pythonocc-core.git
mkdir build
cd build
cmake -DOCE_INCLUDE_PATH=/home/user/source/oce-OCE-0.16.1/build/include  -DOCE_LIB_PATH=/home/user/source/oce-OCE-0.16.1/build/Unix/x86_64-Release-64 ..
make
```

And catch Error
```
[  0%] Building CXX object CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o
/home/user/source/pythonocc-core/build/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx:4585:32: fatal error: Standard_Failure.hxx: No such file or directory
 #include <Standard_Failure.hxx>
                                ^
compilation terminated.
CMakeFiles/_AIS.dir/build.make:62: recipe for target 'CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o' failed
make[2]: *** [CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/_AIS.dir/all' failed
make[1]: *** [CMakeFiles/_AIS.dir/all] Error 2
Makefile:137: recipe for target 'all' failed
make: *** [all] Error 2
```

How I can get working **liboce-ocaf-dev oce-draw** in PPA only 0.15 and 0.17 version?

http://ppa.launchpad.net/freecad-maintainers/oce-release/ubuntu/pool/main/o/oce/

aothms, 2015-09-14:
AFAICT the include files are one directory level deeper. Try invoking cmake instead with:

    -DOCE_INCLUDE_PATH=/home/user/source/oce-OCE-0.16.1/build/include/oce

Hope that helps,

Kind regard,
Thomas

tpaviot, 2015-12-08:
@Valber Did @aothms answer help ?

Valber, 2015-12-08:
No

```
unzip OCE-0.16.1.zip 
cd oce-OCE-0.16.1/
cmake-gui .
cmake .
cmake-gui .
make -j2
cd ..
git clone https://github.com/tpaviot/pythonocc-core.git
cd pythonocc-core/
mkdir build
cd build/
cmake -DOCE_INCLUDE_PATH=/home/user/source/oce-OCE-0.16.1/include/oce  -DOCE_LIB_PATH=/home/user/source/oce-OCE-0.16.1/Unix/x86_64-Release-64 ..
make
[  0%] Swig source
Scanning dependencies of target _AIS
[  0%] Building CXX object CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o
/home/valber/source/pythonocc-core/build/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx:4585:32: fatal error: Standard_Failure.hxx: No such file or directory
 #include <Standard_Failure.hxx>
                                ^
compilation terminated.
CMakeFiles/_AIS.dir/build.make:62: recipe for target 'CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o' failed
make[2]: *** [CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/_AIS.dir/all' failed
make[1]: *** [CMakeFiles/_AIS.dir/all] Error 2
Makefile:137: recipe for target 'all' failed
make: *** [all] Error 2

 lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 15.04
Release:	15.04
Codename:	vivid

```

jf---, 2015-12-09:
@Valber ,

you took @aothms comment too literal, 

__ /home/user/source/oce-OCE-0.16.1/include/oce__

So you should replace the suggested path to the __actual__ one on your machine.
So by /home/user/source/oce-OCE-0.16.1 the path to the unzipped oce-OCE-0.16.1 directory is meant

Valber, 2015-12-09:
Sorr bro
I thought nessesary is described in INSTALL.md file and was tested

**INCLUDE_PATH=/home/user/source/oce-OCE-0.16.1**
```
$ cd ../pythonocc-core/
$ cmake -DOCE_INCLUDE_PATH=/home/user/source/oce-OCE-0.16.1  -DOCE_LIB_PATH=/home/user/source/oce-OCE-0.16.1/Unix/x86_64-Release-64 ..
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/source/pythonocc-core/cmake-build
$ make
[  0%] Swig source
Scanning dependencies of target _AIS
[  0%] Building CXX object CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o
/home/user/source/pythonocc-core/cmake-build/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx:4585:32: fatal error: Standard_Failure.hxx: No such file or directory
 #include <Standard_Failure.hxx>
                                ^
compilation terminated.
CMakeFiles/_AIS.dir/build.make:62: recipe for target 'CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o' failed
make[2]: *** [CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/_AIS.dir/all' failed
make[1]: *** [CMakeFiles/_AIS.dir/all] Error 2
Makefile:137: recipe for target 'all' failed
make: *** [all] Error 2

$ locate Standard_Failure.hxx
/home/user/source/oce-OCE-0.16.1/inc/Handle_Standard_Failure.hxx
/home/user/source/oce-OCE-0.16.1/inc/Standard_Failure.hxx
```

Trying **INCLUDE_PATH=/home/user/source/oce-OCE-0.16.1/inc**
```
$ cmake -DOCE_INCLUDE_PATH=/home/user/source/oce-OCE-0.16.1/inc  -DOCE_LIB_PATH=/home/user/source/oce-OCE-0.16.1/Unix/x86_64-Release-6
4 ..
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/source/pythonocc-core/cmake-build

[cmake-build]$ make
[  0%] Swig source
Scanning dependencies of target _AIS
[  0%] Building CXX object CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o
In file included from /home/user/source/oce-OCE-0.16.1/inc/Standard_Failure.hxx:10:0,
                 from /home/user/source/pythonocc-core/cmake-build/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx:4585:
/home/user/source/oce-OCE-0.16.1/inc/Standard.hxx:13:36: fatal error: Standard_DefineAlloc.hxx: No such file or directory
 #include <Standard_DefineAlloc.hxx>
                                    ^
compilation terminated.
CMakeFiles/_AIS.dir/build.make:62: recipe for target 'CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o' failed
make[2]: *** [CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/_AIS.dir/all' failed
make[1]: *** [CMakeFiles/_AIS.dir/all] Error 2
Makefile:137: recipe for target 'all' failed
make: *** [all] Error 2
$ locate Standard_DefineAlloc.hxx
/home/user/source/oce-OCE-0.16.1/include/Standard_DefineAlloc.hxx
/home/user/source/oce-OCE-0.16.1/src/Standard/Standard_DefineAlloc.hxx
```
Trying **INCLUDE_PATH=/home/user/source/oce-OCE-0.16.1/include**
```
$ cmake -DOCE_INCLUDE_PATH=/home/user/source/oce-OCE-0.16.1/include  -DOCE_LIB_PATH=/home/user/source/oce-OCE-0.16.1/Unix/x86_64-Release-64 ..
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/source/pythonocc-core/cmake-build
$ make
[  0%] Swig source
Scanning dependencies of target _AIS
[  0%] Building CXX object CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o
/home/user/source/pythonocc-core/cmake-build/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx:4585:32: fatal error: Standard_Failure.hxx: No such file or directory
 #include <Standard_Failure.hxx>
[... 35 lines cut ...]
HAVE_LIMITS_H should be defined"
 #  error "check oce-config.h file or compilation options: OCE_HAVE_CLIMITS, OCE_HAVE_LIMITS, or OCE_HAVE_LIMITS_H should be defined"
    ^
In file included from /home/user/source/oce-OCE-0.16.1/inc/Standard_OStream.hxx:22:0,
                 from /home/user/source/oce-OCE-0.16.1/inc/Standard_ExtCharacter.hxx:28,
                 from /home/user/source/oce-OCE-0.16.1/inc/Standard_PrimitiveTypes.hxx:43,
                 from /home/user/source/oce-OCE-0.16.1/inc/Handle_Standard_Transient.hxx:27,
                 from /home/user/source/oce-OCE-0.16.1/inc/Handle_Standard_Failure.hxx:17,
                 from /home/user/source/oce-OCE-0.16.1/inc/Standard_Failure.hxx:16,
                 from /home/user/source/pythonocc-core/cmake-build/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx:4585:
/home/user/source/oce-OCE-0.16.1/inc/Standard_Stream.hxx:43:4: error: #error "check config.h file or compilation options: either OCE_HAVE_IOSTREAM or OCE_HAVE_IOSTREA
M_H should be defined"
   #error "check config.h file or compilation options: either OCE_HAVE_IOSTREAM or OCE_HAVE_IOSTREAM_H should be defined"
    ^
In file included from /home/user/source/oce-OCE-0.16.1/inc/Standard_Failure.hxx:29:0,
                 from /home/user/source/pythonocc-core/cmake-build/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx:4585:
/home/user/source/oce-OCE-0.16.1/inc/Standard_SStream.hxx:35:27: fatal error: strstream.h: No such file or directory
     #include <strstream.h>
                           ^
compilation terminated.
CMakeFiles/_AIS.dir/build.make:62: recipe for target 'CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o' failed
make[2]: *** [CMakeFiles/_AIS.dir/Unix/x86_64-MinSizeRel-64/AISPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/_AIS.dir/all' failed

```

jf---, 2015-12-10:
hi @Valber ,

Looks like you didnt grasp the previous comment...

here, **/home/user/source/oce-OCE-0.16.1/build/include/oce** is a fictional path... we dont know where the source is located on your machine... please make sure the directory your pointing to is the right one as pointed out in earlier comments.

Finally, I suggest you install pythonocc from conda, since building a CAD kernel is not a task if you're new to compiling software.

[Grab it here](https://anaconda.org/DLR-SC/pythonocc-core/) I think you'll have more fun delving into pythonocc this way.

Valber, 2015-12-12:
Hi @jf---
 
>Finally, I suggest you install pythonocc from conda, since building a CAD kernel is not a task if you're new to compiling software.
 
>Grab it here I think you'll have more fun delving into pythonocc this way.
 
Sorry, but on my computer intalled OCE 0.17 for FreeCAD (from ppa).
 
Python OCC needed 0.16 OCE,
 
I Download OCE 0.16.1 https://github.com/tpaviot/oce/releases/tag/OCE-0.16.1
 
In cmake-gui activate
```
OCE_ADD_HEADERS
OCE_COPY_HEADERS_BUILD
OCE_DRAW
```
oce compiled OK
 
After compilation some *.hxx files put in inc/ - directory , other into include/ dir.
Simple copying, don't work, but
 
sudo checkinstall
 
thats build package with
 
```
dpkg -c oce-16.deb
 
-rw-r--r-- root/root   1182496 2015-12-09 00:04 ./usr/local/lib/libTKLCAF.so.9.0.0
-rw-r--r-- root/root     96216 2015-12-09 00:06 ./usr/local/lib/libTKBinTObj.so.9.0.0
-rw-r--r-- root/root    953440 2015-12-09 00:28 ./usr/local/lib/libTKCAF.so.9.0.0
...
-rw-r--r-- root/root       910 2014-11-30 08:46 ./usr/local/include/oce/Handle_SelectMgr_DataMapNodeOfDataMapOfObjectSelectors.hxx
-rw-r--r-- root/root      1998 2014-11-30 08:46 ./usr/local/include/oce/GeomFill_DegeneratedBound.hxx
-rw-r--r-- root/root      1907 2014-11-30 08:46 ./usr/local/include/oce/ShapeSchema_PColgp_SeqNodeOfHSequenceOfVec.hxx
-rw-r--r-- root/root      3923 2014-11-30 08:46 ./usr/local/include/oce/BRepLib_FindSurface.hxx
-rw-r--r-- root/root      3181 2014-11-30 08:46 ./usr/local/include/oce/TColStd_DataMapNodeOfDataMapOfIntegerListOfInteger.hxx
...
```
After install this oce package
 
Build pythonocc with
```
cmake -DOCE_INCLUDE_PATH=/usr/local/include/oce/  -DOCE_LIB_PATH=/usr/local/lib/ ..
make
sudo checkinstall
```
 
Build ok
pythonocc.deb

```
dpkg -c build/pythonocc_20151210-1_amd64.deb

-rw-r--r-- root/root     39653 2015-12-10 10:21 ./usr/lib/python2.7/dist-packages/OCC/Geom2dAPI.py
-rw-r--r-- root/root    469296 2015-12-10 10:07 ./usr/lib/python2.7/dist-packages/OCC/_BOPTools.so
-rw-r--r-- root/root    158568 2015-12-10 10:05 ./usr/lib/python2.7/dist-packages/OCC/_AdvApprox.so
-rw-r--r-- root/root      8269 2015-12-10 10:05 ./usr/lib/python2.7/dist-packages/OCC/ApproxInt.py
-rw-r--r-- root/root     61016 2015-12-10 10:27 ./usr/lib/python2.7/dist-packages/OCC/IntAna.py 
```
 
After installation, check example http://www.pythonocc.org/quick-examples/hello-dumb-box/
 
And catch this error https://github.com/tpaviot/pythonocc-core/issues/86
 
But after that 
```
echo $PATH|sed 's/:/\n/g'
/usr/local/sbin
/usr/local/bin
/usr/sbin
/usr/bin
/sbin
/bin
/usr/games
/usr/local/games
/usr/local/include/oce
/usr/local/lib
/usr/lib/python2.7/dist-packages/OCC
 
python step.py
Traceback (most recent call last):
  File "step.py", line 2, in <module>
    from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox
  File "/usr/lib/python2.7/dist-packages/OCC/BRepPrimAPI.py", line 32, in <module>
    _BRepPrimAPI = swig_import_helper()
  File "/usr/lib/python2.7/dist-packages/OCC/BRepPrimAPI.py", line 28, in swig_import_helper
    _mod = imp.load_module('_BRepPrimAPI', fp, pathname, description)
ImportError: dynamic module does not define init function (init_BRepPrimAPI)
 
```
 

Is it possible to install python-occ from pip, and change his OCE search path?

jf---, 2016-02-17:
@tpaviot , shall we close this?

closed 2016-02-18 by tpaviot