tpaviot/pythonocc-core #146: segfault on OCC.BRepGProp.BRepGProp_Face.Normal()

aothms, 2015-10-20:
Hi,

Following code segfaults for me:

Kind regards,
Thomas

    import OCC.BRepMesh, OCC.TopExp, OCC.TopAbs, OCC.TopoDS, OCC.BRep, OCC.TopLoc, OCC.BRepGProp, OCC.gp
    from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox
    box = BRepPrimAPI_MakeBox(10, 10, 10).Shape()

    def tupelize(xyz):
        return xyz.X(), xyz.Y(), xyz.Z()

    mesh = OCC.BRepMesh.brepmesh_Mesh(box, 1)
    exp = OCC.TopExp.TopExp_Explorer(box, OCC.TopAbs.TopAbs_FACE)
    positions, normals, indices = [], [], []
    while exp.More():
        face = OCC.TopoDS.topods_Face(exp.Current())
        loc = OCC.TopLoc.TopLoc_Location()
        tri_handle = OCC.BRep.BRep_Tool_Triangulation(face,loc)
        tmp_pnt, tmp_vec = OCC.gp.gp_Pnt(), OCC.gp.gp_Vec()
        if not tri_handle.IsNull():
            trsf = loc.Transformation()
            tri = tri_handle.GetObject()
            nodes = tri.Nodes()
            uvs = tri.UVNodes()
            prop = OCC.BRepGProp.BRepGProp_Face(face)
            for i in range(1, nodes.Length() + 1):
                positions.extend(tupelize(nodes.Value(i).Transformed(trsf)))
                uv = uvs.Value(i)
                prop.Normal(uv.X(), uv.Y(), tmp_pnt, tmp_vec) # <-- segfault

aothms, 2015-10-20:
The following, more succinct code does as well:

    import OCC.gp, OCC.BRepGProp, OCC.BRepBuilderAPI

    pln = OCC.gp.gp_Pln()
    face = OCC.BRepBuilderAPI.BRepBuilderAPI_MakeFace(pln).Face()
    prop = OCC.BRepGProp.BRepGProp_Face(face)
    p, v = OCC.gp.gp_Pnt(), OCC.gp.gp_Vec()
    prop.Normal(0., 0., p, v) # <-- segfault

aothms, 2015-10-20:
It seems one can use the following as a work-around:

    import OCC.gp, OCC.BRepLProp, OCC.BRepBuilderAPI, OCC.BRepAdaptor_Surface

    pln = OCC.gp.gp_Pln()
    face = OCC.BRepBuilderAPI.BRepBuilderAPI_MakeFace(pln).Face()
    adapter = OCC.BRepAdaptor.BRepAdaptor_Surface(face)
    prop = OCC.BRepLProp.BRepLProp_SLProps(adapter, 0., 0., 1, 1.e-3)
    prop.Normal()

jf---, 2015-10-20:
Hi Thomas,

I could not reproduce, but that the point / vector is passed by-reference is a little suspicious

```python
In [1]: %cpaste
Pasting code; enter '--' alone on the line to stop or use Ctrl-D.
:import OCC.gp, OCC.BRepGProp, OCC.BRepBuilderAPI
:
:pln = OCC.gp.gp_Pln()
:face = OCC.BRepBuilderAPI.BRepBuilderAPI_MakeFace(pln).Face()
:prop = OCC.BRepGProp.BRepGProp_Face(face)
:p, v = OCC.gp.gp_Pnt(), OCC.gp.gp_Vec()
:prop.Normal(0., 0., p, v) # <-- segfault
:--

In [2]: p
Out[2]: <OCC.gp.gp_Pnt; proxy of <Swig Object of type 'gp_Pnt *' at 0x109885450> >

In [3]: v
Out[3]: <OCC.gp.gp_Vec; proxy of <Swig Object of type 'gp_Vec *' at 0x1098854b0> >

In [5]: from OCC import VERSION

In [6]: VERSION
Out[6]: '0.16.1-dev'

In [7]: 
```

jf---, 2015-12-09:
@aothms , can we close this issue?

aothms, 2015-12-10:
Yeah, let's do that. I'll have a look some time later what happened exactly. If I'll ever figure out I'll reopen.

closed 2015-12-10 by aothms