tpaviot/pythonocc-core #147: Crash with the module libTKMath.9.dylib on examples/core_load_step_ap203.py

smor, 2015-10-20:
Hi!

I'm having some trouble using pythonOCC on my Macbook Air with OS X 10.11 El Capitan.
I compiled pythonocc-core 0.16 and the matching OCE 0.16 from source. Everything went smoothly.
I then tried several examples, with various levels of success.

When I ran `./python core_load_step_ap203.py`, though, it first crashed in the command line with the following error `Reason: unsafe use of relative rpath libTKernel.9.dylib`.

After quite some research, this was solved by [disabling System integrity Protection](https://www.quora.com/How-do-I-turn-off-the-rootless-in-OS-X-El-Capitan-10-11). Apparently, El Capitan changed the way it deals with libraries, and it won't load libraries from `/usr/local`, [as explained here on Homebrew's Github](https://github.com/Homebrew/homebrew/blob/master/share/doc/homebrew/El_Capitan_and_Homebrew.md#if-usrlocal-does-not-exist).

Now the script seems to work, but it ends with a segmentation fault : 11 and a GUI with this error message "Python a quitté de manière imprévue lors de l’utilisation du module libTKMath.9.dylib.".

```
      ...    Step File Reading : ./models/as1_pe_203.stp
      ...    STEP File   Read    ... 
Elapsed time: 0 Hours 0 Minutes 0.010748 Seconds 
CPU user time: 0 seconds 
CPU system time: 0 seconds 
      ... Step File loaded  ... 
Elapsed time: 0 Hours 0 Minutes 0.016043 Seconds 
CPU user time: 0.01 seconds 
CPU system time: 0 seconds 
   4670 records (entities,sub-lists,scopes), 12176 parameters

      ... Parameters prepared ... Elapsed time: 0 Hours 0 Minutes 0.023999 Seconds 
CPU user time: 0.01 seconds 
CPU system time: 0 seconds 
      ...   Objets analysed  ... 
Elapsed time: 0 Hours 0 Minutes 0.036887 Seconds 
CPU user time: 0.02 seconds 
CPU system time: 0 seconds 
  STEP Loading done : 2881 Entities
Segmentation fault: 11
```

The details of the crash are those:
```
Process:               Python [4609]
Path:                  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               2.7.10 (2.7.10)
Code Type:             X86-64 (Native)
Parent Process:        bash [461]
Responsible:           Terminal [459]
User ID:               501

PlugIn Path:             /usr/local/lib/libTKMath.9.dylib
PlugIn Identifier:       libTKMath.9.dylib
PlugIn Version:          ??? (0)

Date/Time:             2015-10-20 15:39:39.771 +0200
OS Version:            Mac OS X 10.11 (15A284)
Report Version:        11
Anonymous UUID:        C3B17987-51EE-8763-021E-AEE6A3FF9EDD

Time Awake Since Boot: 2600 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000

VM Regions Near 0:
--> 
    __TEXT                 0000000102d25000-0000000102d26000 [    4K] r-x/rwx SM=COW  /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libTKMath.9.dylib             	0x0000000103639c6e BSplCLib::LocateParameter(int, TColStd_Array1OfReal const&, TColStd_Array1OfInteger const&, double, bool, int&, double&) + 62
[... 313 lines cut ...]
===========                     =======  ======= 
TOTAL                            419.6M      604 

Model: MacBookAir6,2, BootROM MBA61.0099.B19, 2 processors, Intel Core i5, 1,4 GHz, 4 GB, SMC 2.13f15
Graphics: Intel HD Graphics 5000, Intel HD Graphics 5000, Built-In
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E384A544D4C41522D4E544D
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80AD, 0x483943434E4E4E384A544D4C41522D4E544D
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x117), Broadcom BCM43xx 1.0 (7.21.94.25.1a3)
Bluetooth: Version 4.4.0f4 16320, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Network Service: USB Ethernet, Ethernet, en3
Serial ATA Device: APPLE SSD SD0128F, 121,33 GB
USB Device: USB 3.0 Bus
USB Device: Card Reader
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: Apple USB Ethernet Adapter
Thunderbolt Bus: MacBook Air, Apple Inc., 23.6
```

I couldn't find anything here of anywhere online, so any help would be greatly appreciated!!
This software looks really good, and I'd love to be able to use this feature.

Best regards,
Stéphane

jf---, 2015-10-20:
hi @smor ,

Thanks for the heads up, also the hint for el capitane is appreciated.
I'd like to understand a little bit more about your setup, my guess is the following:

- you installed OCE from homebrew ( cool OCE is on homebrew btw ) 
- you've build pythonocc with the OCE libs from homebrew

Thing is, that I can't reproduce the issue, hence my interest in how you've set things up, then we can dig a little deeper. The fail happens at a method where we've seen more issues ( BSplCLib::LocateParameter ). There has been an issue building this lib with the latest clang compiler, and this smells a lot like it.

If you can inform me a bit more on your specifics, that would be great.

Thanks,

-jelle

[image]

smor, 2015-10-20:
Hi Jelle,

Thanks for the swift answer!

As the setup was more than a month ago, I just had a doubt, but I remember compiling on the train, fearing for my battery life. Actually, I compiled OCE 0.16 from the source tarball. This is confirmed by my `~/.bash_history` file, which also shows that I compiled OCE without any options, which might explain things…

```
cd oce-OCE-0.16/
mkdir build
cd build/
cmake ..
make -j2
sudo make  install/strip
```
The libs live at `/usr/local/lib/libTK*` which caused the first error with El Capitan.

And pythonocc-core was pulled and built from Github
```
smor@stephane-mor:~/dev/pythonocc-core$ git remote show origin
* remote origin
  Fetch URL: git://github.com/tpaviot/pythonocc-core.git
  Push  URL: git://github.com/tpaviot/pythonocc-core.git
  HEAD branch: master
```

It shows that I'm using HEAD. Is that what I'm supposed to do?

I just tried running the tests:
```
smor@stephane-mor:~/dev/pythonocc-core/test$ python run_tests.py 
[...] every thing ok [...]
test_bezier_surfaces (core_geometry_unittest.TestGeometry)
Test: Bezier surfaces ... Segmentation fault: 11
```

So everything goes fine until the Bezier surfaces test. Does that help?
Bezier surfaces, b-splines, they're close concepts and math related…

I could go on the homebrew route (do you advise it?), but as my use of the software is not time-critical, it's a good opportunity to fix a bug! ;-)

Best regards,
Stéphane

WBB-Badtaste, 2015-12-26:
I have the same problem. The testing crash in test_bezier_surfaces model.

rainman110, 2016-01-21:
This is an issue when compiling OCE with clang 6.0 and newer due to optimizations: https://github.com/tpaviot/oce/issues/576

We fixed in the most recent conda packages. Just download them via

     conda install -c https://conda.anaconda.org/dlr-sc pythonocc-core 

If you want to compile oce by yourself, either
   - use an older clang compiler
   - or use the patch provided under: https://github.com/DLR-SC/tigl-conda/blob/master/oce/oce-nullrefs-fix.patch

@tpaviot This issue can be closed.

closed 2016-02-18 by tpaviot