tpaviot/pythonocc-core #150: Issues with GeomLib.BuildCurve3d

aothms, 2015-11-05:
The following piece of code in Python throws an exception while the equivalent code in C++ works well and provides me with a bspline curve. The Python version throws a Standard_ConstructionError in _GeomLib.geomlib_BuildCurve3d(*args).

Apart from the exception there are some obvious issues:
- the doubles by reference Standard_Real &MaxDeviation, Standard_Real &AverageDeviation will not be copied over if I'm not mistaken. But that still should not cause the Standard_ConstructionError.
- when I tried to manually supply the parameters with default values (OCC.GeomAbs.GeomAbs_C1, 14, 30), I cannot get SWIG to accept my call because it cannot seem to wrap its head around the GeomAbs_Shape enumeration

Any help as always would be appreciated.

Python:

    import OCC.gp, OCC.Geom, OCC.Geom2d, OCC.GeomAdaptor, OCC.Geom2dAdaptor, OCC.Adaptor3d, OCC.GeomLib

    surface = OCC.Geom.Geom_CylindricalSurface(OCC.gp.gp_Ax3(OCC.gp.gp_Pnt(), OCC.gp.gp_Dir(0,0,1)), 300)
    line = OCC.Geom2d.Geom2d_Line(OCC.gp.gp_Pnt2d(0, 0), OCC.gp.gp_Dir2d(1,25))
    adaptor_curve = OCC.Adaptor2d.Handle_Adaptor2d_HCurve2d(OCC.Geom2dAdaptor.Geom2dAdaptor_HCurve(line.GetHandle()))
    adaptor_surface = OCC.Adaptor3d.Handle_Adaptor3d_HSurface(OCC.GeomAdaptor.Handle_GeomAdaptor_HSurface(OCC.GeomAdaptor.GeomAdaptor_HSurface(surface.GetHandle())))
    curve_on_surface = OCC.Adaptor3d.Adaptor3d_CurveOnSurface(adaptor_curve, adaptor_surface)

    h = OCC.Geom.Handle_Geom_Curve()

    OCC.GeomLib.geomlib_BuildCurve3d(1e-9, curve_on_surface, 0, 1, h, 0, 0)

C++

	auto surface = new Geom_CylindricalSurface(gp_Ax3(gp::Origin(), gp_Dir(0,0,1)), 300);
	auto line = new Geom2d_Line(gp_Pnt2d(0, 0), gp_Dir2d(1,25));
	auto adaptor_curve = Handle(Adaptor2d_HCurve2d)(new Geom2dAdaptor_HCurve(line));
	auto adaptor_surface = Handle(Adaptor3d_HSurface)(new GeomAdaptor_HSurface(surface));
	auto curve_on_surface = Adaptor3d_CurveOnSurface(adaptor_curve, adaptor_surface);

	Handle(Geom_Curve) h;
	double a, b;
	GeomLib::BuildCurve3d(1e-9, curveonsurface, 0, 1, h, a, b);

tpaviot, 2015-11-05:
Hi Thomas,

As you know, python cannot handle parameters passed by reference, or it
would not be pythonic. As a consequence, all C++ methods that have one ore
more Standard_Real& or Standard_Integer& or Standard_Bool& in the method
prototype are slightly modified by the wrapper. Instead, the python wrapper
*returns* those values and you don't have to pass any parameter.

The good way to use the BuildCurve3d static function in pythonocc would be
(just look at the last line):

import OCC.gp, OCC.Geom, OCC.Geom2d, OCC.GeomAdaptor,
OCC.Geom2dAdaptor, OCC.Adaptor3d, OCC.GeomLib

surface = OCC.Geom.Geom_CylindricalSurface(OCC.gp.gp_Ax3(OCC.gp.gp_Pnt(),
OCC.gp.gp_Dir(0,0,1)), 300)
line = OCC.Geom2d.Geom2d_Line(OCC.gp.gp_Pnt2d(0, 0), OCC.gp.gp_Dir2d(1,25))
adaptor_curve =
OCC.Adaptor2d.Handle_Adaptor2d_HCurve2d(OCC.Geom2dAdaptor.Geom2dAdaptor_HCurve(line.GetHandle()))
adaptor_surface =
OCC.Adaptor3d.Handle_Adaptor3d_HSurface(OCC.GeomAdaptor.Handle_GeomAdaptor_HSurface(OCC.GeomAdaptor.GeomAdaptor_HSurface(surface.GetHandle())))
curve_on_surface =
OCC.Adaptor3d.Adaptor3d_CurveOnSurface(adaptor_curve, adaptor_surface)

h = OCC.Geom.Handle_Geom_Curve()

a, b = OCC.GeomLib.geomlib_BuildCurve3d(1e-9, curve_on_surface, 0, 1, h)

HTH

Thomas

tpaviot, 2015-11-05:
@aothms Hi Thomas,

As you know, python cannot handle parameters passed by reference, or it would not be pythonic. As a consequence, all C++ methods that have one ore more Standard_Real& or Standard_Integer& or Standard_Bool& in the method prototype are slightly modified by the wrapper. Instead, the python wrapper *returns* those values and you don't have to pass any parameter.

The good way to use the BuildCurve3d static function in pythonocc would be (just look at the last line):
```
import OCC.gp, OCC.Geom, OCC.Geom2d, OCC.GeomAdaptor, OCC.Geom2dAdaptor, OCC.Adaptor3d, OCC.GeomLib

surface = OCC.Geom.Geom_CylindricalSurface(OCC.gp.gp_Ax3(OCC.gp.gp_Pnt(), OCC.gp.gp_Dir(0,0,1)), 300)
line = OCC.Geom2d.Geom2d_Line(OCC.gp.gp_Pnt2d(0, 0), OCC.gp.gp_Dir2d(1,25))
adaptor_curve = OCC.Adaptor2d.Handle_Adaptor2d_HCurve2d(OCC.Geom2dAdaptor.Geom2dAdaptor_HCurve(line.GetHandle()))
adaptor_surface = OCC.Adaptor3d.Handle_Adaptor3d_HSurface(OCC.GeomAdaptor.Handle_GeomAdaptor_HSurface(OCC.GeomAdaptor.GeomAdaptor_HSurface(surface.GetHandle())))
curve_on_surface = OCC.Adaptor3d.Adaptor3d_CurveOnSurface(adaptor_curve, adaptor_surface)

h = OCC.Geom.Handle_Geom_Curve()

a, b = OCC.GeomLib.geomlib_BuildCurve3d(1e-9, curve_on_surface, 0, 1, h)
```

aothms, 2015-11-05:
Argh. I feel like an idiot. Thanks for the quick reply Thomas. My additional zeros for the reference arguments where sent to the continuity parameter causing the Standard_ConstructionError. Works like a charm now :)

Kind regards,
Thomas

closed 2015-11-05 by aothms

tpaviot, 2015-11-05:
You're welcome, this is not clearly documented, you could'nt know about that.

gggstamm, 2018-05-02:
I have a similar issue; enable to get output parameters passed by reference 
example:
`
from OCC.Standard import Standard_Float
from OCC.gp import gp_Pln, gp_Dir, gp_Ax1

gpPlane = gp_Pln(1, 2, 3, 4)

a = Standard_Float
b = Standard_Float
c = Standard_Float
d = Standard_Float

gpPlane.Coefficients(a, b, c, d)

print(a, b, c, d)  # returns :   5 5 5 5    :-(

gpPlaneNorm = gpPlane.Axis()  # type: gp_Ax1
gpPlaneNormDir = gpPlaneNorm.Direction()  # type: gp_Dir

x = Standard_Float
y = Standard_Float
z = Standard_Float
gpPlaneNormDir.Coord(x, y, z)
print('x =', gpPlaneNormDir.X(), '  y =', gpPlaneNormDir.Y(), '  z =', gpPlaneNormDir.Z())

print(x, y, z)  # returns :   5 5 5    :-(
`