tpaviot/pythonocc-core #151: Where is TPrsStd package in new Release Pythonocc?

jonike, 2015-11-13:
Hi, There!
I want to use OCAF applicaton framework to save the TPrsStd_AISPresentation class in my OCAF document. but i can't find _TPrStd.pyd in python site-packages/OCC directory.

Could anyone to give me some suggestions, please?
Thanks.

jonike, 2015-11-13:
By the way, i found some warning in PCDM_StoreStatus. 
/*******
swig/python detected a memory leak of type 'PCDM_StoreStatus *', no destructor found.
******/
And there is no export TDataStd_Name_Set static function in TDataStd.py, you must manual to do it.
Sorry for my poor english.

jf---, 2015-12-27:
@tpaviot , can you please close the issue?

tpaviot, 2015-12-27:
Well, @jonike is right. The OCAF framework is not completely wrapped.

I suggest we keep this issue opened untill it's done. Let's postpone a complete ocaf support to oce-0.17 support.

jonike, 2016-01-04:
Thanks guys. That's impressive! Great  job. I am waiting for good news.

commit cbbb6ac7 (2016-05-03): Added TKCAF package, including TDataXtd, TNaming, TPrsStd and AppStd modules; Fix #151
commit 4a1b7ae8 (2016-05-08): Added TKCAF package, including TDataXtd, TNaming, TPrsStd and AppStd modules; Fix #151
closed 2016-05-09 by tpaviot