tpaviot/pythonocc-core #152: Testing on Debian Jessie

rsm-imap-cc, 2015-11-25:
Hello,

I just find out about your great program and I wanted to ask for some help. I'm trying to install it over Debian Jessie and after installing the dependences and building the package I get the error:

```
Traceback (most recent call last):
  File "1.py", line 1, in <module>
    from OCC.STEPControl import STEPControl_Reader
  File "/usr/lib/python2.7/dist-packages/OCC/STEPControl.py", line 32, in <module>
    _STEPControl = swig_import_helper()
  File "/usr/lib/python2.7/dist-packages/OCC/STEPControl.py", line 28, in swig_import_helper
    _mod = imp.load_module('_STEPControl', fp, pathname, description)
ImportError: dynamic module does not define init function (init_STEPControl)
```

I ignore if this is a problem from my build or if it is a problem of the program. 

For the dependences I installed oce-OCE-0.16.1 (and all the tests was okey), for the other two dependences I took the native from Debian. SWIG 2.0.12 and cmake 3.0.2.

If you need any info just ask for it,
Thanks!

rsm-imap-cc, 2015-11-29:
I did the installation for python 3.4 and it worked!

jf---, 2015-12-01:
hi @rsm-imap-cc ,

Sorry we havent gotten round to respond to your question before you've resolved it yourself!
If you've resolved this question, can I ask you to close the issue? Thanks for the feedback...

-jelle

rsm-imap-cc, 2015-12-01:
I actually didn't close the bug because the problem of python2 may be a bug since I didn't had any problem with the compilation. 

Anyways installing it for python 3.4 do solves the problem :)

closed 2015-12-01 by rsm-imap-cc