tpaviot/pythonocc-core #153: ProjLib_ProjectOnSurface::Load missing

procArchitect, 2015-11-26:
ProjLib_ProjectOnSurface::Load functions missing

jf---, 2015-11-27:
hi @procArchitect ,

ipython / the docstring provides documents on what argument the class expects.
why not use **ProjLib_ProjectOnSurface**?

```ipython

In [3]: ProjLib_ProjectOnSurface?
Init signature: ProjLib_ProjectOnSurface(self, *args)
Docstring:      <no docstring>
Init docstring:
* Create an empty projector.

:rtype: None

* Create a projector normaly to the surface <S>.

:param S:
:type S: Handle_Adaptor3d_HSurface &
:rtype: None
File:           ~/miniconda/lib/python2.7/site-packages/OCC/ProjLib.py
Type:           type
````

procArchitect, 2015-11-27:
Hi, jf!
Thanks for attention!

Excuse me, if I don't understand, english is not my native language, and, in addition, i made very short message, but i want to say: 
  functions Load from C class ProjLib_ProjectOnSurface not implemented in Python wrapper pythonocc-core/src/SWIG_files/wrapper/ProjLib.i

------------
excerptions:
from:
http://dev.opencascade.org/doc/refman/html/class_proj_lib___project_on_surface.html
```
ProjLib_ProjectOnSurface Class Reference
Public Member Functions
 	ProjLib_ProjectOnSurface ()
 	Create an empty projector. More...
 
 	ProjLib_ProjectOnSurface (const Handle< Adaptor3d_HSurface > &S)
 	Create a projector normaly to the surface <S>. More...
 
virtual void 	Delete ()
 
virtual 	~ProjLib_ProjectOnSurface ()
``` 
#### void 	Load (const Handle< Adaptor3d_HSurface > &S)
 	Set the Surface to <S>. To compute the projection, you have to Load the Curve. More...
 
#### void 	Load (const Handle< Adaptor3d_HCurve > &C, const Standard_Real Tolerance)
 	Compute the projection of the curve on the Surface. More...
``` 
Standard_Boolean 	IsDone () const
 
Handle< Geom_BSplineCurve > 	BSpline () const 
```
--------

#### from pythonocc-core/src/SWIG_files/wrapper/ProjLib.i :
```
class ProjLib_ProjectOnSurface {
	public:
		%feature("compactdefaultargs") ProjLib_ProjectOnSurface;
		%feature("autodoc", "	* Create an empty projector.

	:rtype: None
") ProjLib_ProjectOnSurface;
		 ProjLib_ProjectOnSurface ();
		%feature("compactdefaultargs") ProjLib_ProjectOnSurface;
		%feature("autodoc", "	* Create a projector normaly to the surface <S>.

	:param S:
	:type S: Handle_Adaptor3d_HSurface &
	:rtype: None
") ProjLib_ProjectOnSurface;
		 ProjLib_ProjectOnSurface (const Handle_Adaptor3d_HSurface & S);
		%feature("compactdefaultargs") Delete;
		%feature("autodoc", "	:rtype: void
") Delete;
		virtual void Delete ();
		%feature("compactdefaultargs") IsDone;
		%feature("autodoc", "	:rtype: bool
") IsDone;
		Standard_Boolean IsDone ();
		%feature("compactdefaultargs") BSpline;
		%feature("autodoc", "	:rtype: Handle_Geom_BSplineCurve
") BSpline;
		Handle_Geom_BSplineCurve BSpline ();
};
```
---------
#### Load missing
I'm relatively new on GitHub, but I'm try to fix that missing and make pull request

jf---, 2015-11-28:
hi @procArchitect , the OCC dev docs are for OCC version 7.0.0, while the pythonocc-core wrappers are for OCE 0.16, which match up with OCC 6.7.0

That might account for the difference in API...

tpaviot, 2015-12-08:
@jf--- is right, the Load method is availble in CCT6.8.x whereas current pythonocc relies on OCCT6.7.0. The Load method will be available in the next pythonocc-0.17 release.

closed 2015-12-08 by tpaviot