tpaviot/pythonocc-core #155: Bug: TColStd_Array1OfInteger and SetValue() method

jirihnidek, 2015-12-02:
Hi,
it is not possible to use array of integers in PythonOCC. Here is example of code:

```python
import OCC
array = OCC.TColStd.TColStd_Array1OfInteger(1, 2)
array.SetValue(1, 1)
```

It will end with following error:

    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    TypeError: TColStd_Array1OfInteger_SetValue expected 2 arguments, got 3

The error code contrast with documentation string: http://api.pythonocc.org/OCC.TColStd.html#OCC.TColStd.TColStd_Array1OfInteger.SetValue

When you try to use only "two" arguments e.g.: ```array.SetValue(1)```, then it will only set zero value at the index.

I tested array of floats and this type of array works as expected:

```python
import OCC
array = OCC.TColStd.TColStd_Array1OfReal(1, 2)
array.SetValue(1, 3.1415)
```

tpaviot, 2015-12-02:
It runs correctly on my machine. Do you use the latest version ?

jirihnidek, 2015-12-03:
Hi,
I'm sorry. I noticed this behavior at system, where OCC is installed from repo rpm and version of PythonOCC is 0.16.0. With PythonOCC 0.16.2 it works as expected.

closed 2015-12-03 by jirihnidek