tpaviot/pythonocc-core #156: Where is the OCC.Utils.DataExchange.STEP.STEPImporter.ReadFile?

huskier, 2015-12-03:
When I do the command:
from OCC.Utils.DataExchange.STEP import STEPImporter
I've got an error: No modules named Utils.
I know that there is a project called pythonocc-DataExchange, but there is nothing there.

I wonder where is it?

Thanks.

tpaviot, 2015-12-08:
You're right, it was removed from the OCCUtils package and was not moved back to pythonocc-dataexchange. For now, you can import STEP files using the builtin pythonocc class STEPControl_Reader. See core_load_step* examples, for instance https://github.com/tpaviot/pythonocc-core/blob/master/examples/core_load_step_ap203.py

closed 2015-12-08 by tpaviot

jf---, 2015-12-08:
@tpaviot ,

Possible we're fragmenting over too many GH repo's.
What about porting / merging the data-exchange -> OCCUtils?

huskier, 2015-12-09:
@tpaviot 
Thanks.