tpaviot/pythonocc-core #159: Access udate function?

Lancewise, 2015-12-04:
Using the code below, I can load a stl file. I noticed the update=True in the display.DisplayShape(fan_shp, update=True) line. Is there a way to access this update function and add code to it? for example: printing the x, y, z locations on update.

Code:
from OCC.Display.SimpleGui import init_display
from OCC.StlAPI import StlAPI_Reader
from OCC.TopoDS import TopoDS_Shape

stl_reader = StlAPI_Reader()
fan_shp = TopoDS_Shape()
stl_reader.Read(fan_shp, './models/disk_brake.stl')

display, start_display, add_menu, add_function_to_menu = init_display()
display.DisplayShape(fan_shp, update=True)

start_display()

jf---, 2015-12-08:
hi @Lancewise ,

So the "update" arg in display.DisplayShape is that the viewer is redrawn when the shape is added to the viewer. So its an argument not a function.

I'm not sure if I follow your question, if you want to move the location of a shape, use the ".Move" attribute on a TopoDS_*.

tpaviot, 2015-12-08:
@Lancewise The SimpleGui provides basic viewing features. If you need something more advanced, the bast way to proceed is to sublcass the Viewer3d class and overwrite DisplayShape and mouse events. That's something that needs an example I think.

closed 2016-01-22 by tpaviot