tpaviot/pythonocc-core #160: Problems with qtdisplay

Achim66, 2015-12-08:
hello,
i have problems with the qtdisplay. It doesn't matter which example I'm using. The error message is allways the same.

Error Message:

>>> 
Using PyQt4

Traceback (most recent call last):
  File "D:\Installation\OpenCASCADE\PyOCC\pythonocc-core-master\examples\core_classic_occ_bottle.py", line 217, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display()
  File "C:\Python27\lib\site-packages\OCC\Display\SimpleGui.py", line 159, in init_display
    win.canva.InitDriver()
  File "C:\Python27\lib\site-packages\OCC\Display\qtDisplay.py", line 106, in InitDriver
    self._display.Create()
  File "C:\Python27\lib\site-packages\OCC\Display\OCCViewer.py", line 118, in Create
    self.Init(self._window_handle)
TypeError: in method 'Display3d_Init', argument 2 of type 'long'
>>> ================================ RESTART ================================

Sytem Info:

OS: Windows 7 Enterprise 64-Bit
Graphic Card: Gerforce GTX 670
NVIDIA  DRIVER: 355.98 (64-Bit)

Installed Python Software:

Python: python-2.7.10.msi
PyQt4: PyQt4-4.11.4-gpl-Py2.7-Qt4.8.7-x32.exe
PyOCC: pythonOCC-0.16.2-win32-py27.exe 

Does anybody knows the reason for this?
regards 
Achim

jf---, 2015-12-08:
hi Achim,

[this commit](https://github.com/tpaviot/pythonocc-core/commit/36cf660ab94aca51eaa21116e7c67c6311fde538) should have resolved this.

could you check that your module is using this latest version?
thanks for the detailed report...

Achim66, 2015-12-09:
I made the changes qtDisplay.py.
Now it works. Thanks a lot.
I think we can close this.
regards 
Achim

closed 2015-12-09 by Achim66