tpaviot/pythonocc-core #161: BRep_Tool_Triangulation Nodes Length crashes if face is a triangle

jackcaron, 2015-12-08:
if I have a piece of like this:
```
face = topods_Face(ex.Current())
location = TopLoc_Location()
facing = BRep_Tool_Triangulation(face, location).GetObject()

nodes = facing.Nodes()

nbPoints = facing.NbNodes() # <<-- crash
nbPoints = nodes.Length()   # <<-- crash
```

Note that I'm getting this crash from a STEP file while I visit each SOLID and SHELL.

jf---, 2015-12-08:
you might have to test whether a triangulation exists, before visiting it, eg, test whether nodes.IsNull()

jackcaron, 2015-12-08:
I think I tried the *IsNull* function. I also used the *Checked* function and I'm still using *IsAllocated*.

jf---, 2015-12-08:
do you have a more complete example to re-produce the problem?

jackcaron, 2015-12-08:
I can try to find a simple example. The only file that has this bug is from a client, so I can't share it.

I'll look into it and try to make one.

jf---, 2015-12-09:
@jackcaron ,

are you aware of pythonocc's [webgl support](https://github.com/tpaviot/pythonocc-core/blob/master/examples/core_webgl_threejs_torus.py) through threejs?

jackcaron, 2015-12-09:
I've seen the files, never checked them. Is there an example in there to parse STEP with triangles?

jf---, 2015-12-09:
Since it seems your interest is in webgl / that function implements what your after, its sure looking into I think...

STEP by definition should not encode a mesh, but BRep CAD data...
STL / OBJ / Collada are more commonly used for mesh data.

PythonOCC really deals with BRep, not mesh...

tpaviot, 2016-04-27:
Closing the issue. The STEP file may be corrupted.

closed 2016-04-27 by tpaviot