tpaviot/pythonocc-core #162: fuzzy bool ops examples

jf---, 2015-12-09:
for pythonocc-core 0.17+ show [how fuzzy boolean ops work](http://dev.opencascade.org/index.php?q=node/1056)

rainman110, 2015-12-11:
Hi @jf--- 

Is opencascade 6.9.0 already supported by pythonocc-core? There are many cool features is the recent releases. 

From what I've seen is that your porting it corrently to OCE. Is pythonocc-core already progressing to the new versions?

tpaviot, 2015-12-11:
So far pythonocc-core supports oce-0.16.x

Next step is to move up to oce-0.17.x

OCCT6.9.0 has not been merged yet into OCE.

jf---, 2015-12-11:
So a quick summary:

* There is a WIP version of the pythonoccc-generator for OCE 0.17 courtesy of [@colonelzentor](https://github.com/colonelzentor/pythonocc-generator/commits/vtk-integration). Don't be distracted by the branch name. Yes, VTK bindings would be cool, but lets focus on getting pythonocc-core working on OCE 0.17.

* There is a OCE branch that matches OCC 6.9.1 courtesy of @blobfish. See this [PR on OCE ](https://github.com/tpaviot/oce/pull/587). I've tried it, works ;)

jf---, 2015-12-11:
Oh, I havent gotten the VTK integration to work, admittedly...
This is more related to building VTK than anything else.
Conda is now more updated ( last time I tried only vtk 5 was there, we need vtk 6.2+ )

rainman110, 2015-12-11:
@jf---  Cool! Thanks for letting me now :+1:

jf---, 2016-02-18:
alas, this feature is OCC 6.8.1, OCE 0.17.1 -> OCC 6.8.0

jf---, 2016-09-26:
hi,

so pythonocc 017 does wrap `BOPAlgo_MakerVolume`, so that's exciting 🏆 

```python
mv = BOPAlgo_MakerVolume()
mv.AddArgument(block_cylinder_shape)
mv.SetRunParallel(True) # -> Missing method
mv.SetIntersect(True)
mv.SetFuzzyValue(0.1) # -> Missing method
```

Looking at [the example here](https://dev.opencascade.org/index.php?q=node/1056), a few other methods are missing:

```c++
BRepAlgoAPI_Fuse aFuse;
TopTools_ListOfShape L1, L2;
L1.Append(S1);
L2.Append(S2);
aFuse.SetArguments(L1);
aFuse.SetTools(L2); // -> Missing method
aFuse.SetFuzzyValue(Fuz);
aFuse.Build(); // -> Missing method
```

I think Fuzzy bools are a big deal of OCCT 6.8.1, I'm really stoked on porting some examples.
Bool ops have been something of an achilles heel in OCC, so this is really something ;)

tpaviot, 2016-09-26:
@jf--- pythonocc-0.17 relies on oce-0.17 <=> OCCT6.8.0. OCCT6.8.1 was nevers released, you have to wait for oce/pythonocc to support OCCT69x. I think this ticked can be tagged as invalid.

jf---, 2016-09-26:
Aiii! Ok, hence the oce-0.18 tag... 
Sorry to raise this confusion for the 2nd time...

tpaviot, 2016-10-12:
PR #339 adds the SetRunParallel method to BOPAlgo_Maker and  the Build method to BRepAlgoAPI_Fuse

tpaviot, 2017-04-28:
Commit c11f790 adds a fuzzy boolean operation example (resurrected old emmenthaler example)

[image]

closed 2017-04-28 by tpaviot