tpaviot/pythonocc-core #164: IPython / Jupyter renderer (feature)

rainman110, 2015-12-11:
Hi together,

I once played around with a ipython renderer. This is the result:
https://www.youtube.com/watch?v=uJnsopuIWR4

I could share the (still slow) code with the community, if there is any interest.

The code is based on three.js and an improved ```Tesselator``` class of pythonocc-core.

tpaviot, 2015-12-11:
Hi @rainman110 Interesting. Thomas (@aothms) already contributed a similar code. Maybe you could work on that basis to improve what's already available ?
See https://github.com/tpaviot/pythonocc-contrib/tree/master/WebServer

rainman110, 2015-12-11:
Hi @tpaviot . I've already seen that. This is also quite interesting.

The beauty of ipython though is, that you have a python<->javascript bridge, that e.g. allows you to modify your plots afterwards (see the autorotate feature).

The current implementation from @aothms opens up a static webpage, that is rendered inside a windows of the current page. This way, there is no connection between the ipython widget and the python code - which does not allow these briding features.

Such bridging features could be:
  - enable wireframe by code _after_ the plot is already rendered
  - change the size of the plot
  - export plot to file (screenshot)
  - ...

tpaviot, 2015-12-11:
Does your implementation allow these features ?

rainman110, 2015-12-11:
Not all of them (only size change and autorotate, also setting a different shape afterwards). But these features are possible, as I am using the python<->javasript bridge. The widget is in fact rendered into the current web page.

tpaviot, 2015-12-11:
Feel free to push your code here, I'm interested in both contributions (iPython and improved tesselator). What exactly did you improved in the tesselator ?

rainman110, 2015-12-11:
I added just two things:
  - swig interface to retrieve vertices and normals
  - compute also boundaries (edges) of all faces (can be seen in the video)

aothms, 2015-12-11:
Hi, indeed interesting additions. I would like to point out that the initial viewer I contributed is not static. A tornado webserver is launched in the background that remains fully responsive. For example what is already possible is adding additional topods_shapes for visualisation when the iframe is already rendered. This integration is however not very ipythonic as also mentioned by @jf--- in https://github.com/tpaviot/pythonocc-contrib/pull/2#issuecomment-129845890 . Especially this is still open for improvement imho.

rainman110, 2015-12-11:
@aothms I did not mean static in a sense that there is no 3D interaction. I meant static more in a sense, that the html / javascript code is static/fixed. Maybe it isn't - this was just me impression by flying over the code - which I admittedly did not very long.

Actually I really liked your implementation, as I could add several shapes and it just works. The only "criticism" I had is, that it is indeed not very ipythonic. Maybe we can make your port ipython or merge our codes together.

Things I implemented were:
  - computation of clipping planes (using bounding spheres) to prevent clipping when zooming out. 
  - fitToSceen (aka fitAll), wich fits the scene into the current window

aothms, 2015-12-12:
That would be good, I will probably have time in the new year to revisit this. One of my aims was not to make it ipython specific. That's why in the end an additional tornado web-server is fired up in the background, working just as well on a vanilla python. I have some additional changes myself laying around. Screenshot [1]. Navigation is rather poor in mine. It seems a lot of our effort is complementary, let's try and prevent fragmentation and try to integrate it. +1 on your Tesselator changes btw. that required a bit of a hack in my code.

[1] https://twitter.com/aothms/status/656846626250715136

rainman110, 2015-12-15:
I tried porting my old ipython renderer to jupyter. As it turns out, it's API is changing a lot and I could not achieve something that works.

There are actually two projects out there, that are interesting from the technical point of view:
  - Mdtraj (http://mdtraj.org/latest/)
  - IMolecule (http://patrick-fuller.com/imolecule/)

Both use ipython 3d widgets. My code was inspired by mdtraj, which uses ipython traitlets (i.e. the python to javascript bridge). Although this project is actively developed, they also have problems with the newest jupyter api.

IMolecule takes a similar approach as @aothms . It has basically two options:
 1. Start a tornado web server (like @aothms does in his widget). This allows rendering into a separate web page/viewer.
 2. Embed static html code directly into the ipython page, which allows an inline rendering into the notebook.

The downside is, that some of the cool features that are available with the python-javascript bridge are not available. On the other side, it should work also with jupyter api changes as it only uses a very small part of the api.

milkowski, 2016-01-10:
Hi, this is very ineresting concept, I want to add my 2 cents to this effort. Look at https://github.com/yhat/rodeo This is Jupyter based, scientific Python IDE on top of Electron platform (http://electron.atom.io). I think it could become state of the art IDE for PythonOCC also, especially together with CadQuery extensions https://github.com/dcowden/cadquery. I wonder if that would be OpenSCAD killer in the future.

jf---, 2016-01-12:
hi @milkowski , those are some interesting ideas...
that said, the references mentioned by @rainman110 are more to the point in getting a "respectable", idiomatic ipython viewer for pythonocc. 

I think that's also the right place to start for loftier ambitions :sunglasses:

tpaviot, 2018-01-25:
Feature available in the master branch

closed 2018-01-25 by tpaviot