tpaviot/pythonocc-core #166: Speed up travis build

rainman110, 2015-12-11:
Currently, OCE is beeing built before each pythonocc-core built.

This does not seem to be necessary. Instead, we could load the conda packages of OCE. 

This would give a significant speedup with respect to built times.

jf---, 2015-12-11:
:+1: 

yep, this is waisting a lot of cycles to get the same binary.
to make this work is [there was earlier discussion here](https://github.com/tpaviot/pythonocc-contrib/issues/10)

for this to work well, I think we need to have in place:

  * automated versioning

  * at that point, for instance using [python-versioneer](https://blog.mozilla.org/warner/2012/01/31/version-string-management-in-python-introducing-python-versioneer/) could make sense. a pythonocc-core version string would be something like  **major.minor.patch.sha1**, eg **0.16.3.cf186a**

  * a commit to OCE/master should result in a new conda package: __0.16.3.the_latest_sha1_of_oce_master__ and would also need to use automated versioning

rainman110, 2015-12-11:
@jf--- We don't necessarily need the versioning. If we want a fixed oce version (0.16.1), this is already possible. The current build also just downloads oce 0.16.1 and builds it.

Thus, we could replace the oce build process with

    conda install oce=0.16.1

tpaviot, 2015-12-12:
@rainman110 Yes, pythonocc compilation first compiles OCE each time a new build is triggered. I used to downoad oce binaries with apt-get, but freecad mainteners team, who provided this binary, switched up to oce-0.17 on their ppa. It was easier for me to build oce each time than setting up my own ppa on launchpad. I will see how to include conda downloads into travis.

rainman110, 2015-12-12:
By the way, we (DLR) also have linux repositories for OCE 0.17, see here:
https://build.opensuse.org/package/show/home:martinsiggel/OCE

The opensuse build system also provides a nice download page:
https://software.opensuse.org/download.html?project=home%3Amartinsiggel&package=OCE

Unfortunately, we can't use the packages for pythonocc-core, but they might come handy for some of your users as you only provide binaries for windows.

tpaviot, 2016-01-22:
travis now uses conda to build pythonocc-core on both osx and linux platforms

closed 2016-01-22 by tpaviot