tpaviot/pythonocc-core #168: When I try to run the sample, I got a problem.

WBB-Badtaste, 2015-12-21:
wubinbindeMacBook-Air:cmake-build wubinbin$ sudo python /Users/wubinbin/Downloads/pythonocc-core/examples/core_display_export_to_image.py 
Traceback (most recent call last):
  File "/Users/wubinbin/Downloads/pythonocc-core/examples/core_display_export_to_image.py", line 21, in <module>
    from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox
  File "/Library/Python/2.7/site-packages/OCC/BRepPrimAPI.py", line 32, in <module>
    _BRepPrimAPI = swig_import_helper()
  File "/Library/Python/2.7/site-packages/OCC/BRepPrimAPI.py", line 28, in swig_import_helper
    _mod = imp.load_module('_BRepPrimAPI', fp, pathname, description)
ImportError: dlopen(/Library/Python/2.7/site-packages/OCC/_BRepPrimAPI.so, 2): Library not loaded: libTKAdvTools.9.dylib
  Referenced from: /Library/Python/2.7/site-packages/OCC/_BRepPrimAPI.so
  Reason: unsafe use of relative rpath libTKAdvTools.9.dylib in /Library/Python/2.7/site-packages/OCC/_BRepPrimAPI.so with restricted binary

How can I fix it?

jf---, 2015-12-21:
are you on osx 10.11?
this sounds like something to do with the new [SIP](https://reverse.put.as/2015/10/12/rootfool-a-small-tool-to-dynamically-disable-and-enable-sip-in-el-capitan/) feature...

osx is becoming too much like windows if you ask me...
[image]

WBB-Badtaste, 2015-12-21:
I have colsed the SIP. I think it's not the reason.

WBB-Badtaste, 2015-12-21:
If I wanna reinstall the OX, which version should I choice?

jf---, 2016-03-29:
@WBB-Badtaste where are your [oce](https://anaconda.org/dlr-sc/oce) libs installed?

WBB-Badtaste, 2016-04-06:
Oh, It's working well now. THX.

closed 2016-04-06 by tpaviot