tpaviot/pythonocc-core #170: SetPixelFormat failed. Error code: 2000

Rafa2016, 2015-12-23:
Hi,
I get the following stacktrace when initializing the GUI:

Traceback (most recent call last):
  File "ooo.py", line 224, in <module>
    window.ui.viewer.InitDriver()
  File "C:\Python27\lib\site-packages\OCC\Display\pyqt4Display.py", l
InitDriver
    self._display.Create()
  File "C:\Python27\lib\site-packages\OCC\Display\OCCViewer.py", line
eate
    self.Init(self._window_handle)
RuntimeError: Aspect_GraphicDeviceDefinitionError
OpenGl_Window::CreateWindow: SetPixelFormat failed. Error code: 2000

I use the 0.16.0 version of pythonOOC on Windows 7.
I saw this same error mentioned here:
https://github.com/tpaviot/pythonocc-core/issues/76
But still have the error even if I have the most recent graphic card driver and I tried an older pythonocc version.
Looking forward for your help and support
Thanks.

Regards
Rafik

jf---, 2015-12-23:
Hi @Rafa2016 ,

So 0.16.2 is the most recent release.
It would be interesting for us to know which backend your using ( pyqt, pyside, wx? )
However, the RuntimeError hints that this issue stems from OCC, the library pythonocc wraps.

Could you please inform us what kind of hardware your running on?
Thanks for mentioning #76... that's appreciated...

Rafa2016, 2015-12-23:
Hi @jf--- ,
Thanks for your response.
Actually I'm using PyQt v4.11.4 for python v 2.7.
I'm running the application on a ThinkPad T410 laptop:
Graphic card: Intel(R) HD Graphics with the most recent  driver.

regards
Rafik

jf---, 2015-12-23:
Thanks @Rafa2016  for the details, that helps...
Issue #76 is also related to an intel HD card...
I think this is about as much support as we can provide...

Most likely wedging into [this discussion on the OCC forum](http://www.opencascade.com/content/chrash-intel-hd-graphics) is the best advice I can give you, knowing that's probably not what you'd like to hear...

Rafa2016, 2015-12-23:
Hi @jf--- ,
Thanks for your support.
The line found in the OCCViewer.py file was:
self.View.TriedronDisplay(Aspect_TOTP_RIGHT_LOWER, Quantity_NOC_BLACK, 0.1, V3d_ZBUFFER), I commented it out, but unfortunately it has not resolved the issue.

Rafik

jf---, 2015-12-23:
Aiii...
Sorry, I'm afraid I'm out of ideas, other than what mentioned before, reporting to the OCC forum...
Sorry to hear so @Rafa2016 ...

Rafa2016, 2015-12-23:
Thank you so much @jf---  for your help anyway, I look forward to finding the right solution.

Regards
Rafik

rainman110, 2016-01-11:
Dear @Rafa2016, did you use windows? 

If yes, please upgrade to the most recent version 0.16.3 through conda. It has probably something to do with the CSF_GraphicShr variable, which i fixed in the current release.

Rafa2016, 2016-01-11:
Dear @rainman110 ,
Thank you very much for your reply, I've just upgraded to the most recent version 0.16.3 from this link:
https://github.com/tpaviot/pythonocc-core/releases.
But I still have the error, please let me know if I am missing something or if something should be done.

rainman110, 2016-01-11:
Please uninstall the installed pythonocc-core. Instead install it instead with __conda__ and try again. Install instructions can be found here:

https://github.com/tpaviot/pythonocc-core/blob/master/INSTALL.md#install-with-conda

You need to download miniconda first in case you don't have it already.

Rafa2016, 2016-01-14:
Hi @rainman110 ,
Thanks for your reply,  before applying your suggestion, I tried to install the library on a new laptop,
Version 0.16.0:
I get another error:
  File "ooo.py", line 513, in <module>
    window.ui.viewer.InitDriver()
  File "C:\Python27\lib\site-packages\OCC\Display\pyqt4Display.py", line 80, in InitDriver
    self._display.Create()
  File "C:\Python27\lib\site-packages\OCC\Display\OCCViewer.py", line 105, in Create
    self.Init(self._window_handle)
RuntimeError: Aspect_DriverDefinitionError
Version 0.16File "C:\Python27\lib\site-packages\OCC\Display\qtDisplay.py", line 31, in <module>
    QtCore, QtGui, QtWidgets, QtOpenGL = get_qt_modules()
  File "C:\Python27\lib\site-packages\OCC\Display\backend.py", line 214, in get_qt_modules
    raise ValueError("no backend has been imported yet with "
ValueError: no backend has been imported yet with ``get_backend``....3: 
Question;
pythonOCC-0.16.X win32 that what you mean by pythonocc-core?
 I have noticed also that the fileshas been renamed thraugh the versions(0.16.0 to 0.16.3):
pyqt4Display To qtDisplay and also others files.

rainman110, 2016-01-15:
Dear @Rafa2016 ,

the project was renamed to pythonocc-core a while ago. This is the reason, why the __conda package__ is called pythonocc-core.

The windows installers offered under the _Releases_ tab suffer from some problems. The main problem is, that they are not properly relocatable. The main problems are hard-coded build/install paths inside the OCE libraries. Another problem is, that the installer does not install  packages for the visualization (e.g. pyqt).

Conda takes care of these issues and fixes them. Conda is a platform-independent package manager for all kind of software (not only python, but mainly focused on python packages).

To install pythonocc-core via conda, you have to download miniconda first. You can get it here:
http://conda.pydata.org/miniconda.html

Then open a command prompt and type:

     conda install -c https://conda.anaconda.org/dlr-sc pythonocc-core 

This will install pythonocc-core and all its dependencies (including python, pyqt and oce). Your mentioned problem above was, that pyqt was not installed on the new laptop. With conda, it gets automatically installed.

It is always important, to have the most recent version installed (0.16.3 currently) as many problems are ironed out in each release. Conda can automatically update packages after you installed them once.

Rafa2016, 2016-01-15:
Hi @rainman110 ,
When I have uninstalled Python OCC and remove the OCC folder from C:\Python27\Lib\site-packages, then I install python-core from conda by executing the commande you have mentionned above,the OCC folder is installed on this folder: C:\Users\lenovo\Miniconda2\Lib\site-packages and not inside the python folder.
So when I execute the application I get this error:

rainman110, 2016-01-15:
It seems, that you are calling the wrong python interpreter. Use the python interpreter from conda:

      C:\Users\lenovo\Miniconda2\python.exe

Rafa2016, 2016-01-15:
Firstly, when I launch this interpreter and put then this line: import OCC.Display.pyqt4Display
I get the same error: ImportError: No module named pyqt4Display.
My application is located on another folder, How can I call this interpreter to execute my application.

rainman110, 2016-01-15:
try this:
   
     import OCC.Display.qtDisplay

rainman110, 2016-01-15:
Please have a look onto the pythonocc examples, e.g. core_helloworld.py . Try to run this first.

Rafa2016, 2016-01-15:
this is the error that I get:
>>> import OCC.Display.qtDisplay
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\lenovo\Miniconda2\lib\site-packages\OCC\Display\qtDisplay.py", line 31, in <module>
    QtCore, QtGui, QtWidgets, QtOpenGL = get_qt_modules()
  File "C:\Users\lenovo\Miniconda2\lib\site-packages\OCC\Display\backend.py", line 214, in get_qt_modules
    raise ValueError("no backend has been imported yet with "
ValueError: no backend has been imported yet with ``get_backend``...

rainman110, 2016-01-15:
Can you please post the output of

     conda list

There should be pyqt4 in the list. It seems, that it isn't though.

Rafa2016, 2016-01-15:
This is the output of the commande:
conda list
# packages in environment at C:\Users\lenovo\Miniconda2:
#
conda                     3.19.0                   py27_0    defaults
conda-env                 2.4.5                    py27_0    defaults
freeimage                 3.17.0                    vc9_0  [vc9]  dlr-sc
freetype                  2.5.5                     vc9_0  [vc9]  defaults
gl2ps                     1.3.8                     vc9_0  [vc9]  dlr-sc
jpeg                      8d                        vc9_0  [vc9]  defaults
libpng                    1.6.17                    vc9_1  [vc9]  defaults
libtiff                   4.0.6                     vc9_1  [vc9]  defaults
menuinst                  1.3.2                    py27_0    defaults
msvc_runtime              1.0.1                     vc9_0  [vc9]  defaults
numpy                     1.10.1                   py27_0    defaults
oce                       0.16.1                    vc9_5  [vc9]  dlr-sc
openssl                   1.0.2e                    vc9_0  [vc9]  defaults
pip                       7.1.2                    py27_0    defaults
pycosat                   0.6.1                    py27_0    defaults
pycrypto                  2.6.1                    py27_3    defaults
pyqt                      4.11.4                   py27_4    defaults
python                    2.7.11                        0    defaults
pythonocc-core            0.16.3               py27_vc9_2  [vc9]  dlr-sc
pyyaml                    3.11                     py27_2    defaults
qt                        4.8.7                     vc9_5  [vc9]  defaults
requests                  2.9.1                    py27_0    defaults
setuptools                19.2                     py27_0    defaults
sip                       4.16.9                   py27_2    defaults
tbb                       4.3.6                     vc9_0  [vc9]  dlr-sc
wheel                     0.26.0                   py27_1    defaults
zlib                      1.2.8                     vc9_2  [vc9]  defaults

rainman110, 2016-01-15:
Are you sure are only import qtDisplay and do nothing else?

This error message occurs only, if you call qtViewer3d::initDriver. 

@jf---  can you have a look at this? I think this is your business here :smirk:

Rafa2016, 2016-01-15:
Yes, I'm sure, I launched the python interpreter from conda as you told me from Miniconda folder, and then I put the import line:import OCC.Display.qtDisplay, nothing else.

rainman110, 2016-01-15:
Hi @Rafa2016 ,

I am sorry, this bahaviour is normal. This was a recent change just before the release. First, one has to open the backend via ```get_backend```. Please have a look onto the OCC.Display.SimpleGui.py / core_helloword.py example.

Rafa2016, 2016-01-15:
Sorry, I didn't get what are you expecting now, could you explain me the issue please?

Rafa2016, 2016-01-19:
Hi @rainman110 ,
This is the output when I launch SimpleGui.py : 
INFO:OCC.Display.backend:backend loaded: qt-pyqt4
INFO:__main__:GUI backend set to: qt-pyqt4
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Traceback (most recent call last):
  File "C:\Python27\Lib\site-packages\OCC\Display\SimpleGui.py", line 165, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display()
  File "C:\Python27\Lib\site-packages\OCC\Display\SimpleGui.py", line 143, in init_display
    win.canva.InitDriver()
  File "C:\Python27\lib\site-packages\OCC\Display\qtDisplay.py", line 123, in InitDriver
    self._display.Create()
  File "C:\Python27\lib\site-packages\OCC\Display\OCCViewer.py", line 118, in Create
    self.Init(self._window_handle)
RuntimeError: Aspect_DriverDefinitionError
Le module spÚcifiÚ est introuvable.

rainman110, 2016-01-20:
@Rafa2016  You copied the conda package into you own python path. This is the reason.

The conda package stores information, where to find the OpenGL driver. Since you moved the package, it can't load it anymore.

__TLDR: Don't move conda packages! Use the conda python interpreter instead.__

Rafa2016, 2016-01-20:
@rainman110 :
This is the Conda folder: C:\Users\Rafik\Miniconda, and this is the python folder :C:\Python27.
Could you please explain me how I moved the packages?

rainman110, 2016-01-20:
@Rafa2016  Your conda package is obviously located under C:\Python27\lib\site-packages\OCC\ or this is an old install of pythonocc.

To use the correct python, use C:\Users\Rafik\Miniconda\python.exe

Rafa2016, 2016-01-20:
This is the interpreter that I use :C:\Users\Rafik\Miniconda\python.exe.
Do you mean that I must to remove the OCC folder :C:\Python27\lib\site-packages\OCC\ ?

rainman110, 2016-01-20:
It is strange, that this python interpreter looks under C:\Python27\lib\site-packages.

It's best to remove or rename the OCC directory to force the conda build

Rafa2016, 2016-01-20:
Ok, Now when I put this line import OCC.Display.qtDisplay into this interpreter, I get this error:
ImportError: No module named qtDisplay.

rainman110, 2016-01-20:
Does this work?

    import OCC

What is inside the folder C:\Users\Rafik\Miniconda\Lib\site-packages\OCC\Display ?

What's version is inside C:\Users\Rafik\Miniconda\Lib\site-packages\OCC\__init__.py?

Did you try out the examples that are in the examples directory of pythonocc-core?

Rafa2016, 2016-01-20:
-import OCC work.
-There is no OCC folder inside : C:\Users\Rafik\Miniconda\Lib\site-packages.
-There is no OCC__init__.py.
-How can I find the examples directory of pythonocc-core?

rainman110, 2016-01-20:
Then you should reinstall pythonocc-core. 

Enter

      conda install -c https://conda.anaconda.org/dlr-sc pythonocc-core 

on the terminal.

After that, check that the OCC folder exists.

You can get the examples from out github page. Just clone the source and you'll have them.

Rafa2016, 2016-01-20:
it's done, now when I put this line: import OCC.Display.qtDisplay I get this error:
    raise ValueError("no backend has been imported yet with "
ValueError: no backend has been imported yet with ``get_backend``...

rainman110, 2016-01-20:
Didn't we have this conversation already? See https://github.com/tpaviot/pythonocc-core/issues/170#issuecomment-171819069

This is normal behaviour. You have to set the backend first. 

Did you try the examples? Have a look into these, they show how it works.

I am sorry to be a bit harsh now, but unless you can not confirm, that the core_helloworld example is working on you laptop, I refuse to help you any further. I've said several times, that you should __try the examples first !__

Rafa2016, 2016-01-20:
Ok, I'll try to run the core_helloworld example, now do I must run this from Miniconda?
I am very sorry to bother you, but I'm a little bit confused.

rainman110, 2016-01-20:
1. Download the pythonocc code from github.
2. goto the examples directory
3. run ```python core_helloworld.py``` using condas python interpreter

Rafa2016, 2016-01-20:
It's done, and then I get this error:
Traceback (most recent call last):
  File "C:\Users\lenovo\Downloads\pythonocc-core-master\pythonocc-core-master\examples\core_helloworld.py", line 21, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display()
  File "C:\Users\lenovo\Miniconda2\lib\site-packages\OCC\Display\SimpleGui.py", line 143, in init_display
    win.canva.InitDriver()
  File "C:\Users\lenovo\Miniconda2\lib\site-packages\OCC\Display\qtDisplay.py", line 123, in InitDriver
    self._display.Create()
  File "C:\Users\lenovo\Miniconda2\lib\site-packages\OCC\Display\OCCViewer.py", line 118, in Create
    self.Init(self._window_handle)
RuntimeError: Aspect_DriverDefinitionError
Le module spÚcifiÚ est introuvable.

rainman110, 2016-01-20:
This seems to be on a different computer! WHat about the computer you were using before. It looks like you might be using an old conda package.

Can you first check, if the following file exists:

   C:\Users\lenovo\Miniconda2\Library\bin\TKOpenGl.dll

If yes, can you please enter try this before running the example
 
    set CSF_GraphicShr=C:\Users\lenovo\Miniconda2\Library\bin\TKOpenGl.dll

in the command line. After that run the example from the exact same command line window.

What version of pythonocc do you have on the computer. Please post output of 

    conda list

Rafa2016, 2016-01-20:
-After seeting the line (set CSF_Graphic....)The core_helloworld.py work perfectly...
-yes, it's a different laptop and the file exists.
-Now when I try to run my app, I get the error:
Traceback (most recent call last):
  File "C:\Users\lenovo\Desktop\CAPPTURN3\ooo.py", line 6, in <module>
    import OCC.Display.qtDisplay
  File "C:\Users\lenovo\Miniconda2\lib\site-packages\OCC\Display\qtDisplay.py", line 31, in <module>
    QtCore, QtGui, QtWidgets, QtOpenGL = get_qt_modules()
  File "C:\Users\lenovo\Miniconda2\lib\site-packages\OCC\Display\backend.py", line 214, in get_qt_modules
    raise ValueError("no backend has been imported yet with "
ValueError: no backend has been imported yet with ``get_backend``...

rainman110, 2016-01-20:
Normally it should not be required to set this variable as this is done in the pythonocc code. Unless, this variable has been set before in the windows system.

Please look into the code ```src/addons/Display/SimpleGui.py``` how to use the qtDisplay module now. It is important to set the backend before importing the qtDisplay module.

rainman110, 2016-01-20:
@tpaviot  This issue can be closed. There is no issue and this thread is developing into a support chat.

closed 2016-01-20 by tpaviot

Rafa2016, 2016-01-20:
Sorry @rainman110 , I didn't get what are asking for.
This is the line that I use :import OCC.Display.qtDisplay.
So I must change this line and see how the module has been imported from th SimpleGui.py?

Rafa2016, 2016-01-20:
Hi @rainman110 & @tpaviot : 
My application works now perfectly, I made some modification into my file using the get_backend function.
Thanks and sorry for the inconvenience.

rainman110, 2016-01-20:
Good to hear!

Reshmakonjari112, 2016-05-04:
@Rafa2016  Can please help me out. I'm getting this error.

 File "C:\Miniconda2\lib\site-packages\OCC\Display\backend.py", line 125, in get_backend
    raise ImportError(msg)
ImportError: a backend is already loaded...``get_backend`` can only be called once

tpaviot, 2016-05-04:
@Rafa2016 this bug has been recently fixed. It will soon be released in the 0.16.5 coming release.

Rafa2016, 2016-05-04:
Hi @Reshmakonjari112 ,
According to @tpaviot, the issue will be resolved int the coming release.
This is how I resolved my bug:
from OCC.Display.backend import get_backend
get_backend("qt-pyqt4")
import OCC.Display.qtDisplay
You can contact me if the you still have the error.
Rafik

Reshmakonjari112, 2016-05-05:
Thank you for your reply @tpaviot  and @Rafa2016 .
@Rafa2016  still I'm getting the same error after adding get_backend("qt-pyqt4").

Rafa2016, 2016-05-05:
@Reshmakonjari112 ,
Please post your code source here, so I can have a look at the issue.